Ultimaker/CuraEngine #602: Algorithm for choosing best connected face of an edge while there are more than two faces connected. (in mesh.cpp)

jason26a, 2017-09-06:
double dot = n0 * n1;
        double det = n * n0.cross(n1);
        double angle = std::atan2(det, dot);
        if (angle < 0) angle += 2*M_PI; // 0 <= angle < 2* M_PI

        if (angle == 0)
        {
            cura::logDebug("Overlapping faces: face %i and face %i.\n", notFaceIdx, candidateFace);
            if (!has_overlapping_faces)
            {
                cura::logWarning("Mesh has overlapping faces!\n");
            }
            has_overlapping_faces = true;
        }
        if (angle < smallestAngle)
        {
            smallestAngle = angle;
            bestIdx = candidateFace;
        }
    

I would like to know :
1. why to choose best face by smallest angle? 
2. why the best connected face be overlapping face(angle = 0)? aren't there will be miss other correct face?
3. why not choose by angle nearly 2* M_PI?

Many thanks.

Ghostkeeper, 2017-09-11:
1. The angle here is the angle between two normal vectors. We're assuming here that the face which has the most similar normal vector is the proper connected face. So we assume that the surface is smooth, so to say. This sort of detects when two surfaces are intersecting with each other which faces belong to which surface.
2. If it's an overlapping face with the same normal vector, then we assume that it belongs to the same surface.
3. Actually I think there's a small bug here, because if the angle is slightly negative it will be very close to `2 * M_PI` (so it is the "worst choice" according to this code) but the angle of the normal is still very similar to the normal vector of the original face, so in actuality it should be a very good candidate. I think it would be better to subtract `M_PI` from the angle, then make that value absolute (positive) and take the smallest angle of that.

Ghostkeeper, 2017-09-25:
I think you understood the algorithm correctly. It doesn't show the merit of the algorithm though.

What I meant was this case (seen from top-down if you will, to make it 2D):

     A         B
      \       /
       \     /
        \   /
         \ /
          C
         / \
        /   \
       /     \
      /       \
     /         \
    D           E

In this case, the face AC gets linked to the face CE, because they have the same normal vector (in fact, the vector is exactly the same, so `angle == 0` and a warning gets posted). Also, the face BC gets linked to CD. This is the most logical conclusion in a real-world environment: We assume that the surface is continuous rather than that there is a kink in both surfaces.

jason26a, 2017-09-29:
[image]

I would like to make my question more clear..  please see the picture above.
CADB is a surface of a cube, the triangle ABC is connect to triangle ADB.
And ABF is another triangle which overlapping to ABC.

In this case, I think the best triangle connected to ABC will be ADB instead of ABF.
But according to the code : 

FPoint3 n0 = FPoint3(vertices[notFaceVertexIdx].p - vertices[idx0].p).cross(v0);
FPoint3 v1 = vertices[idx_candidate].p - vertices[idx0].p;
FPoint3 n1 = v1.cross(v0);

The normal of ABC is AC X AB, and the normal of ABF is AF X AC so the angle between them
is 0, which means same orientation. and the normal of ADB is AD X AB, so ABC and ADB has 
opposite orientation normals.

As your mention above, I think triangle ADB is correct triangle which connected to ABC.
To make the ABC's normal same direction to ABC ( degree ==0) , I think the code should be
modify to:

FPoint3 n1 = v0.cross(v1);

Please advice.

Ghostkeeper, 2017-10-04:
Let me get this right by thinking along and writing my thoughts down here.

In your example, connecting vertices to the code:
* `idx0` is A
* `idx1` is B
* `notFaceVertexIdx` is C
* `idx_candidate` is either D or F. Where it goes wrong is D.

We then get:

    v0 = AB
    n0 = AC x AB
    v1 = AD
    n1 = AD x AB

Because the angle direction between AC/AB and AD/AB is reversed, it indeed seems logical that we need to compute `n1 = AB x AD`. In the case of ABF, the angle direction between AC/AB and AF/AB is not reversed so if we invert the cross product then we do get the opposite direction, which is logical too.

So yeah I agree. I get it now. Let's invert that `n1`.

commit 51fda149 (2017-10-04): Fix winding order of normals for face matching

Fixes #602. See also #602 for an explanation of why this is correct.
closed 2017-10-04 by Ghostkeeper