Ultimaker/CuraEngine #615: CuraEngine master branch compile fail under ubuntu linux 64 bit

musex, 2017-09-29:
I already install protobuf-3.4.1 & libArcus, and just follow the instruction, then it report fail. is this a c++11 feature that g++ on my linux not support? I not very familiar with g++, and the g++ in my ubuntu is 4.8.4
$uname -a
Linux iZ2ze66hn1pkopqvbdkpp2Z 3.13.0-86-generic #130-Ubuntu SMP Mon Apr 18 18:27:15 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
$g++ --version
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.8/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.4-2ubuntu1~14.04.3' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3)
--------------------------------------------------------------------
here is the fail detail, hope someone can help me......
$mkdir build &&cd build
$cmake ../
$make
Scanning dependencies of target clipper
[  1%] Building CXX object CMakeFiles/clipper.dir/libs/clipper/clipper.cpp.o
[  2%] Linking CXX static library libclipper.a
[  2%] Built target clipper
[  4%] Running C++ protocol buffer compiler on Cura.proto
Scanning dependencies of target _CuraEngine
[  5%] Building CXX object CMakeFiles/_CuraEngine.dir/src/bridge.cpp.o
[  6%] Building CXX object CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o
In file included from /home/share/oss/CuraEngine/src/pathPlanning/Comb.h:11:0,
                 from /home/share/oss/CuraEngine/src/LayerPlan.h:8,
                 from /home/share/oss/CuraEngine/src/FffGcodeWriter.h:15,
                 from /home/share/oss/CuraEngine/src/FffProcessor.h:5,
                 from /home/share/oss/CuraEngine/src/commandSocket.cpp:7:
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(Args ...)’:
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:18: error: expected ‘,’ before ‘...’ token
             [args...]()
                  ^
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:18: error: expected identifier before ‘...’ token
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:21: error: parameter packs not expanded with ‘...’:
             [args...]()
                     ^
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:21: note:         ‘args’
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:36:34: error: expansion pattern ‘args’ contains no argument packs
                 return new T(args...);
                                  ^
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(const std::function<_Res(_ArgTypes ...)>&, Args ...)’:
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:21: error: expected ‘,’ before ‘...’ token
             [f, args...]()
                     ^
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:21: error: expected identifier before ‘...’ token
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:24: error: parameter packs not expanded with ‘...’:
             [f, args...]()
                        ^
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:24: note:         ‘args’
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:54:36: error: expansion pattern ‘args’ contains no argument packs
                 return new T(f(args...));
                                    ^
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(const std::function<T*(Args ...)>&, Args ...)’:
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:21: error: expected ‘,’ before ‘...’ token
             [f, args...]()
                     ^
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:21: error: expected identifier before ‘...’ token
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:24: error: parameter packs not expanded with ‘...’:
             [f, args...]()
                        ^
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:24: note:         ‘args’
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/home/share/oss/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:70:30: error: expansion pattern ‘args’ contains no argument packs
                 return f(args...);
                              ^
make[2]: *** [CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o] Error 1
make[1]: *** [CMakeFiles/_CuraEngine.dir/all] Error 2
make: *** [all] Error 2

smartavionics, 2017-09-29:
Hi. Yes, I think your g++ is too old. On a Ubuntu 16.04.3 release, the g++ is 5.4.0 and compiles CuraEngine OK. Sorry, I don't know the minimum g++ version required.

musex, 2017-09-29:
Thank you smartavionics ! I upgrade system to ubuntu 16.04.4 release with g++ 5.4.0, and problem solved！

closed 2017-09-29 by musex

Ghostkeeper, 2017-10-04:
For posterity, the minimum GCC version required is 4.9.2.