Ultimaker/CuraEngine #619: [armhf] StringTest fails

thopiekar, 2017-10-04:
https://launchpadlibrarian.net/339703839/buildlog_ubuntu-artful-armhf.cura-engine_1%3A2.7.0~201710041002~rev1660~pkg168~ubuntu17.10.1_BUILDING.txt.gz

onitake, 2017-10-07:
Let me chime in here: It doesn't only fail on armhf, but on **all** 32-bit architectures:
https://buildd.debian.org/status/package.php?p=cura-engine&suite=unstable

We also had to fix the string format for 64-bit ints in writeInt2mm so sprintf would not truncate them to 32 bits:
```
--- a/src/utils/string.h
+++ b/src/utils/string.h
@@ -38,15 +38,15 @@ static inline void writeInt2mm(const int64_t coord, std::ostream& ss)
 {
     constexpr size_t buffer_size = 24;
     char buffer[buffer_size];
-    int char_count = sprintf(buffer, "%d", int(coord)); // convert int to string
+    int char_count = sprintf(buffer, "%" PRId64, coord); // convert int to string
 #ifdef DEBUG
     if (char_count + 1 >= int(buffer_size)) // + 1 for the null character
     {
-        logError("Cannot write %ld to buffer of size %i", coord, buffer_size);
+        logError("Cannot write %" PRId64 " to buffer of size %i", coord, buffer_size);
     }
     if (char_count < 0)
     {
-        logError("Encoding error while writing %ld", coord);
+        logError("Encoding error while writing %" PRId64, coord);
     }
 #endif // DEBUG
     int end_pos = char_count; // the first character not to write any more
```

ChrisTerBeke, 2017-10-09:
Unfortunately we don't support 32 bit systems anymore (https://github.com/Ultimaker/CuraEngine/issues/547).

thopiekar, 2017-10-09:
I know. But wouldn't it be kind, if we could keep this open, so, if someone
from the community wants and  finds a solution, she or he can send a PR.

The issue is still there and not supporting an architecture doesn't mean
that an issue doesn't exist.

Thanks!

onitake, 2017-10-09:
@thopiekar here's a patch for the broken StringTest: https://anonscm.debian.org/cgit/3dprinter/packages/cura-engine.git/tree/debian/patches/1000-64-bit-safe-stringtest.patch

The `writeInt2mm` bug seems to be fixed already, so you shouldn't need the other one.

Would be great if you could test with that one and report your results!

thopiekar, 2017-10-09:
Hey @onitake ,
did you find the patch accidentally or are you working for/on Debian?
I can give this patch a try and if it works prepare a PR 😉 
But you can prepare the PR also on your own.. You actually did the biggest part and found the patch (in case it really works 😋 ).

Thanks!

thopiekar, 2017-10-09:
Created a PR here: https://github.com/Ultimaker/CuraEngine/pull/621

Ultimakers CI will test it automatically and I'll do also in my PPA on all architectures.

thopiekar, 2017-10-09:
Ah, damn. I see the reason why the build fails, but not why the patch work 🙄 
Still have no good eye for C/C++...

thopiekar, 2017-10-09:
😝

onitake, 2017-10-10:
No, not by accident.

Petter (he's a Debian Developer) and I are preparing Debian packages for  all of Cura.
Since CuraEngine is already in Debian, we were able to release earlier than the rest.

When we enabled the unit tests during build, we were hit by failures on the Debian build servers, and Petter prepared a fix for the (obviously incorrect) string formats. %d accepts an `int`, while the argument is an `int64_t`. On amd64 and some other 64-bit architectures, `int` is 64 bits wide, but that is not the case on 32-bit systems. Unfortunately, there is no universal format string for `int64_t`, but most C standard libraries support `PRId64`.

BagelOrb, 2018-03-06:
outdated

closed 2018-03-06 by BagelOrb