Ultimaker/CuraEngine #622: Debian: Using rapidjson

thopiekar, 2017-10-09:
Just found this, while working on #619:

https://anonscm.debian.org/cgit/3dprinter/packages/cura-engine.git/tree/debian/patches/03-system-rapidjson.patch

Ghostkeeper, 2017-10-19:
It says "Path not found" when I open that link. Did the link get outdated?

thopiekar, 2017-10-19:
Could be. Next time I should better create an comment and copy the whole content of it.

thopiekar, 2017-10-29:
https://anonscm.debian.org/cgit/3dprinter/packages/cura-engine.git/tree/debian/patches/2020-system-rapidjson.patch

Ghostkeeper, 2017-11-06:
It seems @onitake built a patch to replace the RapidJSON and ClipperLib libraries that get distributed along with CuraEngine's source code with pre-installed libraries.

I think that building CuraEngine on systems other than Debian would become somewhat harder then though. Perhaps we should opt for a check-first-then-build combination like I've been doing recently for [SFML in SeriousProton](https://github.com/daid/SeriousProton/blob/041b9b1beaf76e03d4b6acf8ceb8ccfb3e1b1fbf/CMakeLists.txt#L75)?

onitake, 2017-11-07:
We renamed that file to better reflect the status of the patch. 2xxx means: only relevant for Debian releases.
I do agree that using system rapidjson and polyclipping may be useful in general. 

But one should keep in mind that this may require additional maintenance work if a library introduces incompatible changes. It may be necessary to specify (or at least document) compatible versions.

closed 2018-03-12 by thopiekar