Ultimaker/CuraEngine #632: Retrive print time & fillament length from CuraEngine

Shtirliz40k, 2017-10-26:
Hello, everyone.

I’m trying to use CuraEngine to get printing time and filament length using terminal only. My problem is that passing exact same set of parameters to desktop Cura and CuraEngine doesn’t return same results.

I have created an Ultimaker Original printer instance and changed some parameters in Cura GUI. Then take all arguments from cura.log and pass them to CuraEngine using -s arguments with values.

Using these arguments, I call CuraEngine manually like below:

`CuraEngine slice -v -j /path/to/cura/definitions/fdmprinter.def.json <various -s parameters, copy-pasted from cura.log> -g -e0 <extruder related -s parameters, also copied from cura.log> -e0 -l /path/to/model -o /dev/null`

CuraEngine returns an output like below (when “-v” flag is passed):

```
;FLAVOR:Marlin
;TIME:7395
;Filament used: 6.66199m
;Layer height: 0.6
End of gcode header.
Print time: 7395
Print time (readable): 2h 3m 15s
Filament: 42499
```

But these values differs from values in Desktop Cura. Relative difference is bigger when printing time is smaller and tends to 0 when models are big.

I double checked whole set of parameters (counting around 200 overall) and they match completely. Seems like Cura changes some input parameters internally and doesn’t show it.

What could cause the issue? How can use CuraEngine so that output is the same with desktop Cura?

Thanks in advance!

Ghostkeeper, 2017-11-06:
Apart from the settings, Cura's front end also modifies the models. For instance, it puts it down on the build plate if it weren't already. Maybe your model is slightly below the build plate?

Shtirliz40k, 2017-11-07:
@Ghostkeeper Thanks for your help. Yes, some of my models really were below the build plate. It looks like CuraEngine ignores a part of model, which is bellow than build plate.

Just two questions left:
- I have found `-s center_object=True` setting for cura engine. Will CuraEngine automatically move my model up, if I pass these parameter?
- Does Cura's frontend make some other changes to models?

Ghostkeeper, 2017-11-11:
> I have found -s center_object=True setting for cura engine. Will CuraEngine automatically move my model up, if I pass these parameter?

Looking at the code, yes apparently it does. I didn't know about this feature. This is the relevant code:

https://github.com/Ultimaker/CuraEngine/blob/b0aba7ffcac314589648c35c6c16c79a2d0aecf0/src/MeshGroup.cpp#L148-L155

So it subtract the minimum Z coordinate of the model from all vertex coordinates.

> Does Cura's frontend make some other changes to models?

It converts the models to triangle lists instead of strips or fans. CuraEngine is only able to handle triangle lists (so 3 vertices per face and no linking of faces to save space). Most STLs you find on the internet will already be triangle lists. The front-end also does the safety checks: Is your model within the printer's volume, are the settings feasible, etc. But you could skip that.

For the rest I don't think the front-end modifies anything.

Shtirliz40k, 2017-11-25:
I have tried to use CuraEngine with `-s center_object=True`, it looks okay. I steel get some results difference, but it is very small, that is unnecessary for me. So, thanks for your help, i think this issue can be closed.

closed 2017-11-25 by Shtirliz40k