Ultimaker/CuraEngine #633: writeTemperatureCommand logic assumes initial extruder number is already set to 0

d235j, 2017-10-26:
The logic at https://github.com/Ultimaker/CuraEngine/blob/84e234dae66e02a9d74ff0ef577d2cb2a6b87ce3/src/gcodeExport.cpp#L1016 tries to be smart and not include the extruder number if the extruder
However, the slicer assumes the extruder is 0. This means that if the printer is already set to a different extruder, the incorrect extruder will be used.

This was introduced in 92937617f59ab5f025d35bf0d55b9dd8328e2674, and there doesn't seem to be a good reason for it. Is there any harm in setting the extruder each time the command is written out?

The alternative to changing this logic would be writing out "T0" (and therefore explicitly setting the extruder to 0) at the beginning of the print.

See @maukcc's comments in https://github.com/Ultimaker/Cura/issues/1128 for some discussion on this issue.

d235j, 2017-10-26:
Likewise, the logic in startExtruder() will not output an` M135 T` command or `T` command if the new_extruder is the same as the current_extruder (including at the beginning since current_extruder is initialized to 0).

Ghostkeeper, 2017-11-06:
Duplicate of https://github.com/Ultimaker/Cura/issues/1819 . Thanks Maukcc.

closed 2017-11-06 by Ghostkeeper