Ultimaker/CuraEngine #634: On enable support [ERROR] Trying to retrieve unregistered setting with no value given: 'support_skip_some_zags'

recarens, 2017-10-27:
This happens only when i'm enabling the support variable : 
"support_enable" : { "default_value": true },

smartavionics, 2017-10-27:
This can happen if the backend is out of sync with the UI. That sometimes happens if you are building from the master branch or using the PPA.

Gigahawk, 2017-10-27:
This happens to me, I built from the CuraEngine-3.0.3 and am using definitions from Cura-3.0.3. Although this seems to happen on the master branch as well.

recarens, 2017-10-30:
I'm working on Cura-3.0.3 to, I created a new def.jason inherits from fdmprinter of this version. is estrange because the parameter 'support_skip_some_zags' is an experimental section...

recarens, 2017-10-30:
In cura 2.7 its working OK the same def.json

closed 2017-10-30 by recarens