Ultimaker/CuraEngine #637: Feature request: hilbert curve top/bottom pattern

bzed, 2017-10-29:
hi,

I know that hilbert curves as top/bottom pattern result in an insane amount of printer movements, but the result just looks great. please implement it like slic3r does :)

https://imgur.com/gallery/SLdeM

thanks,
bernd

Ghostkeeper, 2017-11-06:
You like it just for its aesthetics, do you? It doesn't seem useful in any way and would certainly give quite some trouble with overhang above infill...

You could also use Cross infill at 100% for a door such as that, and remove the top skin so that the infill gets exposed.

Ghostkeeper, 2018-02-05:
Our project manager just removed this feature from our planning because it's older than 12 weeks.

BagelOrb, 2018-03-06:
It might be an option at some point in time, because it might be a side product of the research I am currently doing at Delft Technical University.

Ultimaker won't spend time on this, though.

BagelOrb, 2018-03-29:
Actually I already implemented it, but it will take a while for it to actually appear as an infill option in Cura.

mbartlett21, 2019-06-28:
@BagelOrb When will this be in Cura?

BagelOrb, 2019-06-30:
Well. I've implemented the Hilbert curve itself, but not as an infill option directly. 

I do see how the Hilbert curve might work well for 100% filling with minimal working. It should work better than the CrossFill pattern, since CrossFill doesn't support 100% dense filling. 

However I really don't have the time for this. Perhaps you can find somebody in the Cura team or somebody from outside who would be willing to make the final step from my implementation, which is at https://github.com/Ultimaker/CuraEngine/tree/fractal_dithering

Specifically the file https://github.com/Ultimaker/CuraEngine/blob/fractal_dithering/src/infill/SquareSubdiv.h

MRAAGH, 2021-11-30:
> You like it just for its aesthetics, do you? It doesn't seem useful in any way and would certainly give quite some trouble with overhang above infill...

The hilbert curve is very useful for the initial layer because localizes heat. Neighboring lines (which are all melted simultaneously) merge together, resulting in reliable adhesion even when the nozzle height is not calibrated perfectly.

Virtually any other initial layer pattern is more sensitive to calibration: whenever a fresh line is printed next to a non-melted existing line, the imperfections of the existing line carry over to the new one.

BagelOrb, 2021-11-30:
Interesting perspective. Why do you think that molten lines absorb imperfections better? Overextrusion is overextrusion regardless. If the lines next to the one being printed is still molten then that line will bulge upward, causing the same type of problems, right?

Have you prototyped any of this? Or is it purely theoretical at this point?

MRAAGH, 2021-11-30:
Printed with lines pattern:
[image]

Printed with Hilbert curve, on same bed, same temperatures (Prusa Slic3r):
[image]

1. The first print has artifacts on the first layer. An artifact begins where the bed is higher (I print on glass without auto leveling). If printing long parallel lines, the artifact is emphasized on each subsequent line. What you see is a minor example (I don't have any better ones on me right now). Typically, I end up with an ugly triangular pattern on the underside of the print. Plus a chaotic surface which can cause further complications when printing the second layer.

2. On the first image, notice how the printing lines are very visible? They did not merge together because they were not melted at the same time. Meanwhile, the second print is so smooth you can even see my fingerprint on it. Clearly, such coverage means more contact with the bed, thus better adhesion.

BagelOrb, 2021-12-01:
Very interesting stuff. I wouldn't consider it to be proven yet, though. There are multiple parameters which can influence this.

How many prints did you do with the Hilbert pattern? Does some other slicer offer it?

The Hilbert pattern has a lot of corners, which means that it will be printed at a relatively low speed. Have you compared against just turning the bottom layer speed down on the lines pattern?

zarthcode, 2021-12-01:
Can confirm, Hilbert is a very effective first layer *because of the corners*. A localized area remains hot and there are almost no lifts/retractions/rapids at all. The nozzle isn't far from newly laid plastic, so the adjacent lines haven't cooled - allowing them to have much greater adhesion.  Yes, it's a slow layer, hours on some prints. But because of the pattern, there's almost no opportunity to put down filament next to filament that is cooled.

Slowing down a line pattern is the literal opposite of that. On a corner, you'd be fine, but the longer the line, the more you run up against filament that has cool, and possibly shrunk/pulled/warped.  Hilbert eliminates this at the cost of print time.

BagelOrb, 2021-12-01:
I understand that the HIlbert curve is very localized which causes the adjacent lines to be cooled less compared to all other patterns.

However, I don't really see why that would improve adhesion of the print to the build surface.

That's why I proposed to do the test on slow parallel lines. So that we can judge the speed of the Hilbert pattern separately from its geometry.

zarthcode, 2021-12-01:
> 
> However, I don't really see why that would improve adhesion of the print to the build surface.
> 

Imagine two touching parallel lines on the first layer. Once the first has cooled, it will have a radius on the sides - on the next pass, it won't remelt enough to remove that radius.  On the next, the same thing happens, where they touch, the materials can intermix a bit on contact, but ultimately, there's going to be two radii now - that's a significant gap that increases w/layer height.  A hotter bed and some overextrusion/smushing could produce a similar effect, But it's _still_ going to cool when the nozzle is away.  There will almost always be a texture. If you can feel it, there's a gap.

To simplify it further, imagine two round bars laying side-by-side on a surface. Compare that to two rectangular ones. One you can see under, one you can't. The second has more contact area with each other and with the surface, thus more adhesion.  There's also a lot less room for shrinkage/warping. Scale that across the entire first layer.

> That's why I proposed to do the test on slow parallel lines. 

That's not an apples-to-apples comparison. As I said, it sacrifices speed for localized heating.  It improves adhesion because warping isn't a factor, compared to a line, which can be dragged by the nozzle, etc. 

> So that we can judge the speed of the Hilbert pattern separately from its geometry.

Why? We **know** it's _slooow_.  But it's also _entirely_ dependent upon acceleration/jerk limits of a particular machine.  ...Have you ever used this?  It's not a setting I use for fast prints.  It's for really big/heavy, multiple-day prints that I need to not fail and waste an entire rolls of filament while printing. (But it's a worthy sacrifice to know that when the first layer is down, you're basically through the woods.) It adheres reliably and doesn't facilitate warping in any way.

It's just a tool in the toolbelt - this one just happens to be a big why I maintain a slic3r configuration for my FDM.

BagelOrb, 2021-12-01:
You reasoning makes sense and I am inclined to accept that the Hilbert pattern will indeed improve adhesion.

The comparison I proposed would be to be certain that it's the pattern causing the enhanced adhesion, not the fact that it's printed slow. We know that the Hilbert pattern is slow *and* has a different geometry. In order to be sure that it's the geometry which enhaces adhesion and not the slowness we can make a simple test to see if just adding slowness to the lines pattern improves adhesion already.

zarthcode, 2021-12-01:
I imagine, from that baseline, that it's possible to achieve similar through a pattern that doesn't feature abrupt 90deg turns and constant stops - but still keeps the nozzle in a tight area as to maximize the filament's temperature *after* extrusion.  Something akin to HSM-toolpaths in the CNC world. Tight spiral paths with an option for elongation that could allow the extruder and head to hit higher speeds - the idea being to quickly extrude as much material as possible in as tight an area as permissible.

BagelOrb, 2021-12-02:
Interesting. I think those two are at odds, though. The Hilbert path is the the tightest possible path which covers 2D space evenly with a continuous extrusion line. It might be interesting, though, to deviate from the Hilbert path a bit and define some new pattern which is less local, but prints at higher speeds.

What we should be optimizing is something like the average time between a path and adjacent paths. Any idea on how to do that mathematically? Problem is that in infinite space the average time also goes to infinity.

MRAAGH, 2021-12-02:
> Any idea on how to do that mathematically?

I have an idea: let's cheat! With computers!
Genetic evolution or a similar optimization algorithm could be the way to go. Of course, the problems are as always with genetic algorithms - how do you define the fitness function so it really represents what we're looking for, and how do you define the genome so it's not *too* chaotic.
Alternatively, we could just come up with a bunch of designs on our own and see what works best.

MRAAGH, 2021-12-02:
Here are a few ideas. I like the pink one, it's simple and in theory very effective. If you can think of improvements, don't hesitate to post sketches!
[image]

Ghostkeeper, 2021-12-13:
Could add Cross infill to that mix. It can already be tested in Cura 4.12 by setting the number of bottom layers to 0 and the infill density to 100%. It's a mix of 90 and 45 degree turns. But I think it'll leave more gaps than Hilbert would. It has more places where four 90 degree corners come together (hence the name "Cross" I guess).

I think the [Gosper curve](https://en.wikipedia.org/wiki/Gosper_curve) is a good candidate since it doesn't have any places where many sharp corners join, which could result in a gap. However it would have to print in tiles, and the curve would have to get clipped at the edge of the print, which increases complexity. Although I think the latter would have to happen for any of these curves.

Perhaps an added benefit is that these curves don't have long straight lines, and so reduce the effect of internal stresses as the plastic cools down.

mooklepticon, 2023-03-26:
I'd like to refresh this request. CNC Kitchen mentioned how useful it is in his arc overhangs video. The stresses of the hilbert curve cancel out the warping seen in other infill/skin patterns. 
https://www.cnckitchen.com/blog/arc-overhangs-in-prusaslicer search "hilbert"