Ultimaker/CuraEngine #639: CuraEngine unaffected by LGPL license change?

onitake, 2017-10-31:
I noticed that the other Cura packages (libSavitar, libArcus, Cura, Uranium) were switched to LGPL-3+, but the license of CuraEngine remains at AGPL-3+.

Is this intentional, or will the license be changed later?

Ghostkeeper, 2017-11-07:
This is intentional. The license of CuraEngine wasn't changed. It has many contributors that would need to agree with the change. Not as many on paper as Cura's front-end, but many of the front-end contributors contributed only profiles which don't need licensing.

The most important reason for Ultimaker to change the license to LGPL was to allow third parties to create closed source plug-ins. Since the engine has no plug-in structure this is not applicable.

onitake, 2017-11-14:
Ok.
This clears it up.
Thanks!

closed 2017-11-14 by onitake