Ultimaker/CuraEngine #643: bug/machine-nozzle-size-ineffective

chousemath, 2017-11-02:
Hi CuraEngine team,

I wanted to report what I think may be bug. If it is actually my lack of understanding of how CuraEngine / 3D printing works my apologies in advance!

### Desired outcome
> When editing a printer-specific `def.json` file (here is an [example](https://raw.githubusercontent.com/Ultimaker/Cura/master/resources/definitions/101Hero.def.json)), I would like the machine_nozzle_size attribute to affect the resulting gcode

### Expected outcome
> Assuming that the material diameter has not changed, I would expect the `E-dimension` to change for different machine_nozzle_sizes (because more or less material is being extruded for different nozzle sizes)

```
;LAYER_COUNT:175
;LAYER:0
M107
G0 F3600 X1.2 Y0.00 Z0.3
;TYPE:SKIRT
G1 F1800 X1.24 Y-.789 E0.03941 <-- I expect the `E` value to change
G1 X1.359 Y-1.569 E0.07878 <-- I expect the `E` value to change
G1 X1.557 Y-2.334 E0.1182 <-- I expect the `E` value to change
```

### Observed outcome
After slicing a test STL file with two versions of 101Hero.def.json (version A has machine_nozzle_size 0.4, version B has machine_nozzle_size 2.5), I ended up getting exactly the same gcode. I have pulled directly from the master branch of CuraEngine so I believe I have the latest version. I am currently on Ubuntu 16.04. I have included (a) the command I used along with, (b) the relevant sections of the def.json file in each trial, and (c) the CuraEngine verbose output. I suspect that my defined machine_nozzle_size is being overriden by the default 0.4 machine_nozzle_size. Thank you very much for your help!

### (A)
```
./CuraEngine slice -v -j "test/definitions/101Hero.def.json" -o "test/gcode/test.gcode" -l "test/stl/calicat.stl"
```

### (B)
####*machine_nozzle_size = 0.4*
```json
"machine_nozzle_size": { "default_value": 0.4 }
```

####*machine_nozzle_size = 2.5*
```json
"machine_nozzle_size": { "default_value": 3.5 }
```

### (C)
```
Cura_SteamEngine version DEV
Copyright (C) 2017 Ultimaker

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
Loading test/definitions/101Hero.def.json...
Loading test/definitions/fdmprinter.def.json...
[WARNING] Trying to override unknown setting speed_print.
[WARNING] Trying to override unknown setting speed_wall.
[WARNING] Trying to override unknown setting speed_layer_0.
[WARNING] Trying to override unknown setting retraction_speed.
Loading test/stl/calicat.stl from disk...
Loading test/definitions/fdmextruder.def.json...
loading 'test/stl/calicat.stl' took 0.003 seconds
Loaded from disk in 0.020s
[WARNING]  -s mesh_position_z="0" -s center_object="false" -s speed_ironing="20" -s ironing_only_highest_layer="false" -s ironing_enabled="false" -s wireframe_nozzle_clearance="1" -s wireframe_roof_drag_along="0.8" -s wireframe_roof_fall_down="2" -s wireframe_straight_before_down="20" -s wireframe_strategy="compensate" -s wireframe_top_jump="0.6" -s wireframe_flat_delay="0.1" -s wireframe_flow_flat="100" -s wireframe_printspeed_flat="5" -s wireframe_printspeed_down="5" -s wireframe_printspeed_up="5" -s wireframe_printspeed_bottom="5" -s wireframe_height="3" -s magic_fuzzy_skin_thickness="0.3" -s support_conical_angle="30" -s support_conical_enabled="false" -s spaghetti_flow="20" -s spaghetti_max_height="2" -s spaghetti_max_infill_angle="10" -s spaghetti_infill_enabled="false" -s coasting_min_volume="0.8" -s coasting_volume="0.064" -s coasting_enable="false" -s ironing_pattern="zigzag" -s conical_overhang_enabled="false" -s draft_shield_dist="10" -s draft_shield_enabled="false" -s optimize_wall_printing_order="false" -s relative_extrusion="false" -s magic_mesh_surface_mode="normal" -s support_mesh_drop_down="true" -s mold_roof_height="0.5" -s mold_enabled="false" -s meshfix_union_all_remove_holes="false" -s meshfix_union_all="true" -s raft_base_fan_speed="0" -s raft_interface_fan_speed="0" -s raft_surface_fan_speed="0" -s raft_base_jerk="20" -s raft_interface_acceleration="3000" -s anti_overhang_mesh="false" -s raft_surface_acceleration="3000" -s raft_base_speed="15" -s raft_interface_speed="15" -s raft_base_line_spacing="1.6" -s raft_base_thickness="0.3" -s raft_interface_line_width="0.7" -s raft_interface_thickness="0.15" -s raft_surface_thickness="0.1" -s raft_surface_layers="2" -s support_mesh="false" -s raft_airgap="0.3" -s prime_tower_position_x="200" -s raft_margin="15" -s raft_surface_speed="20" -s z_offset_taper_layers="0" -s draft_shield_height_limitation="full" -s z_offset_layer_0="0" -s skirt_gap="3" -s adhesion_type="brim" -s extruder_prime_pos_y="0" -s prime_blob_enable="true" -s infill_mesh_order="0" -s support_minimal_diameter="3" -s remove_empty_first_layers="true" -s support_tower_diameter="3" -s prime_tower_wipe_enabled="true" -s support_use_towers="true" -s support_roof_pattern="concentric" -s support_bottom_line_distance="0.4" -s support_roof_line_distance="0.4" -s support_interface_skip_height="0.3" -s support_roof_height="1" -s support_bottom_enable="false" -s support_roof_enable="false" -s gradual_support_infill_steps="0" -s mold_width="5" -s support_infill_sparse_thickness="0.1" -s support_join_distance="2" -s support_bottom_stair_step_width="5" -s support_bottom_stair_step_height="0.3" -s support_xy_overrides_z="z_overrides_xy" -s support_xy_distance="0.7" -s wireframe_roof_inset="3" -s dual_pre_wipe="true" -s support_line_distance="2.66" -s support_pattern="zigzag" -s support_angle="50" -s support_type="everywhere" -s support_bottom_extruder_nr="0" -s support_roof_extruder_nr="0" -s infill_hollow="false" -s raft_surface_line_spacing="0.4" -s skirt_line_count="1" -s support_extruder_nr_layer_0="0" -s support_infill_extruder_nr="0" -s support_enable="false" -s cool_lift_head="false" -s cool_min_layer_time="5" -s infill_extruder_nr="-1" -s material_standby_temperature="150" -s magic_fuzzy_skin_enabled="false" -s infill_offset_x="0" -s coasting_speed="90" -s infill_pattern="grid" -s material_extrusion_cool_down_speed="0.7" -s acceleration_topbottom="3000" -s skin_outline_count="1" -s infill_mesh="false" -s print_sequence="all_at_once" -s default_material_print_temperature="210" -s retraction_min_travel="1.5" -s z_seam_y="100" -s xy_offset_layer_0="0" -s material_flow="100" -s skirt_brim_speed="30" -s carve_multiple_volumes="true" -s xy_offset="0" -s skin_angles="[ ]" -s jerk_infill="20" -s cool_fan_speed_0="0" -s ironing_flow="10" -s fill_outline_gaps="false" -s jerk_support_bottom="20" -s support_top_distance="0.1" -s fill_perimeter_gaps="everywhere" -s raft_base_line_width="0.8" -s z_seam_x="100" -s travel_compensate_overlapping_walls_x_enabled="true" -s z_seam_corner="z_seam_corner_inner" -s raft_interface_jerk="20" -s skin_line_width="0.4" -s smooth_spiralized_contours="true" -s wall_0_inset="0" -s support_line_width="0.4" -s top_bottom_pattern="lines" -s cool_fan_speed_max="100" -s wall_0_wipe_dist="0.2" -s raft_surface_line_width="0.4" -s wall_line_width_x="0.4" -s wall_line_count="2" -s machine_gcode_flavor="RepRap (Marlin/Sprinter)" -s prime_tower_line_width="0.4" -s bottom_layers="6" -s magic_fuzzy_skin_point_dist="0.8" -s retraction_count_max="90" -s cool_fan_enabled="true" -s cool_fan_full_layer="2" -s speed_travel="120" -s machine_shape="elliptic" -s z_seam_relative="false" -s jerk_skirt_brim="20" -s roofing_line_width="0.4" -s acceleration_ironing="3000" -s extruder_prime_pos_z="0" -s skin_alternate_rotation="false" -s acceleration_enabled="false" -s wall_line_width_0="0.4" -s top_bottom_pattern_0="lines" -s machine_center_is_zero="true" -s retraction_combing="all" -s cross_infill_pocket_size="2" -s material_final_print_temperature="195" -s raft_base_acceleration="3000" -s switch_extruder_prime_speed="20" -s acceleration_infill="3000" -s acceleration_print_layer_0="3000" -s machine_filament_park_distance="16" -s machine_nozzle_head_distance="3" -s slicing_tolerance="middle" -s machine_acceleration="4000" -s brim_line_count="20" -s skirt_brim_line_width="0. [... cut ...]
[WARNING] There are empty layers (or layers with empty extruder plans) in the print! Temperature control and cross layer travel moves might suffer.
[WARNING] There are empty layers (or layers with empty extruder plans) in the print! Temperature control and cross layer travel moves might suffer.
Progress: export accomplished in 0.422s
Total time elapsed  0.56s.
Gcode header after slicing:
;FLAVOR:Marlin
;TIME:4947
;Filament used: 3.36239m
;Layer height: 0.2
End of gcode header.
Print time: 4947
Print time (readable): 1h 22m 27s
Filament: 8087
```

smartavionics, 2017-11-02:
Err, I don't think the size of the nozzle affects the volume of plastic extruded so the E values won't change. Having said that, the size of the nozzle should influence your choice of line width as quality will suffer if the line width is wildly wrong for the nozzle size. Changing the line width should change the E values.

Ghostkeeper, 2017-11-08:
Indeed, change the line width. When you're using the front-end, changing the nozzle size the line width automatically scales along, but if you just run the engine from command line, it only uses the default_values so the line width stays the same.

The nozzle size is only used in the engine for the wiping on the prime tower and writing the correct nozzle size to g-code headers, so it really won't affect much.

chousemath, 2017-11-08:
Thank you for the advice! I will give changing line_width a try today and come back with my results.

chousemath, 2017-11-08:
Here are my results:

Using the default 101Hero def.json file, I confirmed that CuraEngine is correctly loading in that json file (by observing the command line output) and sliced a STL file. I double-checked that the 101Hero def.json file did not have any preexisting setting for line_width. The default line_width value for CuraEngine is 0.4mm. A small snippet of the generated gcode has been included directly below:

```
;LAYER_COUNT:175
;LAYER:0
M107
G0 F3600 X1.2 Y0.00 Z0.3
;TYPE:SKIRT
G1 F1800 X1.24 Y-.789 E0.03941
G1 X1.359 Y-1.569 E0.07878
G1 X1.557 Y-2.334 E0.1182
G1 X1.831 Y-3.074 E0.15757
G1 X2.179 Y-3.783 E0.19697
G1 X2.597 Y-4.454 E0.23641
G1 X3.08 Y-5.078 E0.27578
```

**I then performed the following tests:**

* add `"line_width": { "default_value": 0.1 }` to the 101Hero def.json file
* add `"wall_line_width": { "default_value": 0.1 }` to the 101Hero def.json file (replacing the line above)
* add `"wall_line_width_x": { "default_value": 0.1 }` to the 101Hero def.json file (replacing the line above)
* add `"wall_line_width_0": { "default_value": 0.1 }` to the 101Hero def.json file (replacing the line above)
* include `-s line_width=0.1` as a command line argument
* include `-s wall_line_width=0.1` as a command line argument (replacing the argument above)
* include `-s wall_line_width_0=0.1` as a command line argument (replacing the argument above)
* include `-s wall_line_width_x=0.1` as a command line argument (replacing the argument above)

In all cases, there seemed to be no difference in the section of gcode included above (I am expecting the E-value to be different). Is there a flaw in the way I am conducting my tests? I'm sure I've missed something important. Thank you again for your time!

smartavionics, 2017-11-08:
How about changing the skirt/brim line width as your example above is showing the start of the skirt gcode.

smartavionics, 2017-11-08:
That's "skirt_brim_line_width"

Ghostkeeper, 2017-11-14:
They are not having an effect because the `default_value` is being overwritten by the `value`. The `value` uses a formula to derive its value from the parent settings.

Try this:

    "wall_line_width": {"value": "0.1"}

BagelOrb, 2017-11-20:
Command line slicing doesn't calculate formula's in the `value` function. CuraEngine only uses the `default_value` values...

If you want the formula's evaluated you need some kind of frontend.

closed 2018-02-27 by BagelOrb