Ultimaker/CuraEngine #669: Feature request: Initial layer line width instead of factor

cangelis, 2017-12-15:
Initial layer line width is configured for bed adhesion most of the time, so defining different line widths for different extrusion types makes the configuration harder. I would prefer to set constant line width for all the extrusion types in the first layer.  For example, I use different line width for support, wall_0 and wall_x.

`initial_layer_line_width_factor` = 190 (because I want 0.8mm extrusion lines for first layer)
`support_line_width` = 0.35 -> 0.35 * 1.9 = 0.66 **too low**
`wall_0` = 0.42 -> 0.42 * 1.9 = 0.8 **expected**
`wall_x` = 0.5 -> 0.5 * 1.9 = 0.95 **too much**

What about defining `initial_layer_line_width` ?

Ghostkeeper, 2018-01-02:
I agree. The different line widths are mostly intended to get better quality for walls while still retaining the speed for infill and support and such. The quality doesn't matter much in the first layer because the layer height is completely different, it gets squashed by the bed, it sags when it melts if you have a heated bed, etc.

It's more useful to specify a fixed line width and easier to understand for the user.

Ghostkeeper, 2018-01-02:
A problem with this though is that the current profiles will change which could have a great effect on the build plate adhesion.

Also, the skirt/brim line width has no effect any more.

BagelOrb, 2018-03-06:
We could do with an Initial Layer Extra Line Width.

That way we can retain the multiplier as well.

New line width = feature_line_width * multiplier + extra_line_width

That way the user can choose the amount of influence the feature type has. You can just set the multiplier to 0% and the Extra_line_width to 0.4mm.

Just an idea...

Ghostkeeper, 2018-03-29:
Hmm, well, while that's powerful, I think that would be too complex to use for people that aren't intimate with the code.

BagelOrb, 2018-03-29:
Aren't almost all settings expert settings, though?

The only settings we want users to understand is the profiles and enabling support etc.

The other settings are there for advanced users to define what a profile looks like.