Ultimaker/CuraEngine #674: Feature request: Ironing on every or/and N layers

megablue, 2017-12-26:
I would like to request Ironing not being exclusive to top-most layer but ob every layer or selected interval.
The idea being, ironing fills the microvoids between lines, ironing on every layer will make the parts to be super strong since the microvoids between extrusion lines are basically eliminated and bond together better.

ChrisTerBeke, 2017-12-29:
Hi, thanks for your feature request, sounds useful. Could you maybe open an issue on the main Cura repository, it's way more active with contributors and get's more attention in general. On this repo we mainly use issues for bug reports for the CuraEngine specifically.

megablue, 2017-12-30:
alright, thanks.

Ghostkeeper, 2018-01-17:
This is a feature request for CuraEngine specifically, right?

Ghostkeeper, 2018-01-17:
Have you tried setting the "Top Surface Skin Line Width" setting smaller? I'd think that would have a similar effect to what you want.

ChrisTerBeke, 2018-01-17:
It requires implementation in the engine indeed, but also UI settings, so the broader feature is more for Cura as a whole, which I think we should keep track of in the main repo.

Ghostkeeper, 2018-01-17:
Nearly all new features of the engine require a setting in the UI, because we can't normally change the default behaviour of the engine without having a test sequence go over the result.

ChrisTerBeke, 2018-01-17:
That's true. Just trying to keep everything together as many people only look at the issues in the main Cura repo, not following all repositories we have. Personally I'd even go as far as removing the ability to file issues here and redirect all to the main repo, but that's just an opinion :)

BagelOrb, 2018-03-06:
The idea does indeed change the structural properties of the print, but at the cost of a lot of print time.

I don't think Ultimaker has any time soon to implement this issue. Other issues are more pressing.

Closing this to clean up the amount of open issues.

closed 2018-03-06 by BagelOrb