Ultimaker/CuraEngine #676: Initial extruder other than T0 causes an unnecessary switch (G-code opens with retraction on unused extruder T0)

stuartpb, 2018-01-01:
If I set up a print that only uses extruder T1 and never uses my T0 extruder at all, the `T1` switch at the beginning of the procedure is preceded with a massive `G1 F1200 E-16` retraction on the initial T0, which makes a *very* unpleasant noise from the stepper motor, as that filament is not able to move with a cooled-off hotend:

```
;FLAVOR:RepRap
;TIME:1074
;Filament used: 0m
;Layer height: 0.15
;Generated with Cura_SteamEngine 3.1.0
T0
M190 S60
M104 T1 S200
M109 S0
M109 T1 S200
M82 ; absolute extrusion mode
; BEGIN START G-CODE
G90 ;  set positioning to absolute
G28 X Y Z ; home XYZ at default_value home feedrates
G92 X0 Y0 Z0 A0 B0 ; define zeroes
M132 X Y Z  ; Recall stored home offsets
; END START G-CODE
M83 ; relative extrusion mode
;LAYER_COUNT:50
;LAYER:0
G1 F1200 E-16 ; <=================== !!!
T1
M109 S200
M107
G0 F1200 X6.54 Y4.837 Z0.3
;TYPE:SKIRT
G1 F600 X6.488 Y4.979 E0.00754
G1 X4.989 Y8.802 E0.20487
G1 X4.949 Y8.899 E0.00523
G1 X3.643 Y11.964 E0.16622
; ...etc...
```

Essentially, [this code](https://github.com/Ultimaker/CuraEngine/blob/caf7dd7b0e38af0018a6c2200bd7a8e66bdd9b63/src/LayerPlan.cpp#L789-L793) needs to recognize that the first extruder isn't one you need to "switch" to.

stuartpb, 2018-01-02:
Indeed, now that I think about it, it seems like this assumption that the active extruder will be 0 will be *even more* of a problem in 3.2, as #640 sets the G-code to start with a T command switching to the *real* initial extruder.

stuartpb, 2018-01-02:
The *other* way to fix this (short of initializing `current_extruder` to the start_extruder_nr in the `GCodeExport` constructor, which looks like it'd require significant refactoring?) would be to initialize `current_extruder` to -1, then add an `else if (current_extruder < 0) {startExtruder(new_extruder); return;}` [here](https://github.com/Ultimaker/CuraEngine/blob/master/src/gcodeExport.cpp#L916) to initialize based on the value of the first `switchExtruder`.

jackha, 2018-01-02:
Devs, see: CURA-4751

Ghostkeeper, 2018-01-17:
This E-16 comes from putting the nozzle into parking position, because Cura assumes that T0 is active, heated up and the filament is in position for printing after the start g-code is completed.

There is an assumption there that T0 is active, which has been brought up before: https://github.com/Ultimaker/Cura/issues/1819 and several other times that I remember. But barring that, I think assuming that the active extruder is heated up and brought in position to print is fair. Instead we'd rather need a way for the printer to say which toolchain is active at the beginning of a print, either via the printer definition file or an active (wifi/USB) connection. But this is much more complex.

Or did I understand your problem wrongly?

stuartpb, 2018-01-17:
Yeah, no, I'm just describing that the problem is the slicer assuming the initial extruder *of the routine* is T0, which isn't necessarily the case (as mentioned [here](https://github.com/Ultimaker/CuraEngine/issues/676#issuecomment-354687366)).

s00500, 2018-05-27:
It would be great if this retraction could be turned off completly since it can be a problem for some printers

diegopradogesto, 2018-05-29:
Our manager removed this from the planning. It won't be fixed soon.

FalconFour, 2019-03-06:
This causes me a problem because I load one extruder with one material, and one with another... and the models I choose to print are using one extruder or the other - not both (I have no need/interest in dual extruding single prints). So when I start a print using only extruder 2, it goes through its start Gcode, then unexpectedly snorts back a whole length of filament and starts printing!

The resulting code:
`...;Put printing message on LCD screen
M117 Printing...
G92 E0
G1 F1800 E-4
;LAYER_COUNT:140
;LAYER:0
G1 F1200 E-16
G92 E0
T1
G92 E0
M107
G0 F3600 X119.408 Y64.625 Z0.3
...`
See how my start Gcode ends (after "Printing..."), then it goes on and makes an E-4 call, then an E-16? I'd been trying to debug why the printer primes itself (with my start Gcode), but then retracts and prints an empty brim for almost a whole minute. :(

No way to avoid this?

Ghostkeeper, 2019-03-06:
The Search-and-Replace post-processing script always works for that sort of thing.

I'm not able to reproduce that though. Here's the start of the g-code I get when slicing with a Custom FDM Printer using only the second extruder:
```
;FLAVOR:Marlin
;TIME:625
;Filament used: 0m, 0.111376m
;Layer height: 0.1
;Generated with Cura_SteamEngine master
T1
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:99
;LAYER:0
M107
G0 F3600 X39.623 Y39.349 Z0.3
;TYPE:SKIRT
G1 F1500 E0
G1 F1800 X40.222 Y38.834 E0.01486
G1 X40.87 Y38.383 E0.02971
G1 X41.561 Y37.999 E0.04458
...```

It immediately switches to T1. Nothing done on T0. It resets the E a few times too many but that should not make a difference to the printer. Here's a project file: 
[CFFFP_cube.zip](https://github.com/Ultimaker/CuraEngine/files/2935269/CFFFP_cube.zip)

Maybe you could share a project file? It looks like something in your configuration is still doing something with the first extruder.

DIMI-PIMI69, 2020-10-29:
Hello everyone. Could anyone find the answer to this question?

Ghostkeeper, 2020-11-02:
The last comment here requests a project file that reproduces the issue, DIMI-PIMI69. Can you provide such a project file?

tommylux, 2021-01-03:
I have noticed this, not sure what is causing this issue.. thanks