Ultimaker/CuraEngine #677: 为什么 安装了libArcus和Protobuf 编译时还是说报错找不到

winyluo, 2018-01-02:
lib_CuraEngine.a(commandSocket.cpp.o)：在函数‘google::protobuf::internal::GetEmptyStringAlreadyInited()’中：
commandSocket.cpp:(.text._ZN6google8protobuf8internal27GetEmptyStringAlreadyInitedEv[_ZN6google8protobuf8internal27GetEmptyStringAlreadyInitedEv]+0x5)：对‘google::protobuf::internal::fixed_address_empty_string’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘protobuf_Cura_2eproto::protobuf_AssignDescriptors()’中：
Cura.pb.cc:(.text+0xaf8)：对‘google::protobuf::internal::AssignDescriptors(std::string const&, google::protobuf::internal::MigrationSchema const*, google::protobuf::Message const* const*, unsigned int const*, google::protobuf::MessageFactory*, google::protobuf::Metadata*, google::protobuf::EnumDescriptor const**, google::protobuf::ServiceDescriptor const**)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘protobuf_Cura_2eproto::AddDescriptorsImpl()’中：
Cura.pb.cc:(.text+0xbb9)：对‘google::protobuf::MessageFactory::InternalRegisterGeneratedFile(char const*, void (*)(std::string const&))’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘cura::proto::Object::MergePartialFromCodedStream(google::protobuf::io::CodedInputStream*)’中：
Cura.pb.cc:(.text+0x4af0)：对‘google::protobuf::internal::WireFormatLite::ReadBytes(google::protobuf::io::CodedInputStream*, std::string*)’未定义的引用
Cura.pb.cc:(.text+0x4b31)：对‘google::protobuf::internal::WireFormatLite::ReadBytes(google::protobuf::io::CodedInputStream*, std::string*)’未定义的引用
Cura.pb.cc:(.text+0x4b72)：对‘google::protobuf::internal::WireFormatLite::ReadBytes(google::protobuf::io::CodedInputStream*, std::string*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘cura::proto::Object::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const’中：
Cura.pb.cc:(.text+0x4d05)：对‘google::protobuf::internal::WireFormatLite::WriteBytesMaybeAliased(int, std::string const&, google::protobuf::io::CodedOutputStream*)’未定义的引用
Cura.pb.cc:(.text+0x4d46)：对‘google::protobuf::internal::WireFormatLite::WriteBytesMaybeAliased(int, std::string const&, google::protobuf::io::CodedOutputStream*)’未定义的引用
Cura.pb.cc:(.text+0x4d87)：对‘google::protobuf::internal::WireFormatLite::WriteBytesMaybeAliased(int, std::string const&, google::protobuf::io::CodedOutputStream*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘cura::proto::Polygon::MergePartialFromCodedStream(google::protobuf::io::CodedInputStream*)’中：
Cura.pb.cc:(.text+0x81f9)：对‘google::protobuf::internal::WireFormatLite::ReadBytes(google::protobuf::io::CodedInputStream*, std::string*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘cura::proto::Polygon::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const’中：
Cura.pb.cc:(.text+0x83f4)：对‘google::protobuf::internal::WireFormatLite::WriteBytesMaybeAliased(int, std::string const&, google::protobuf::io::CodedOutputStream*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘cura::proto::PathSegment::MergePartialFromCodedStream(google::protobuf::io::CodedInputStream*)’中：
Cura.pb.cc:(.text+0xacf3)：对‘google::protobuf::internal::WireFormatLite::ReadBytes(google::protobuf::io::CodedInputStream*, std::string*)’未定义的引用
Cura.pb.cc:(.text+0xad34)：对‘google::protobuf::internal::WireFormatLite::ReadBytes(google::protobuf::io::CodedInputStream*, std::string*)’未定义的引用
Cura.pb.cc:(.text+0xad75)：对‘google::protobuf::internal::WireFormatLite::ReadBytes(google::protobuf::io::CodedInputStream*, std::string*)’未定义的引用
Cura.pb.cc:(.text+0xadb6)：对‘google::protobuf::internal::WireFormatLite::ReadBytes(google::protobuf::io::CodedInputStream*, std::string*)’未定义的引用
Cura.pb.cc:(.text+0xadf4)：对‘google::protobuf::internal::WireFormatLite::ReadBytes(google::protobuf::io::CodedInputStream*, std::string*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘cura::proto::PathSegment::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const’中：
Cura.pb.cc:(.text+0xaf7e)：对‘google::protobuf::internal::WireFormatLite::WriteBytesMaybeAliased(int, std::string const&, google::protobuf::io::CodedOutputStream*)’未定义的引用
Cura.pb.cc:(.text+0xafbf)：对‘google::protobuf::internal::WireFormatLite::WriteBytesMaybeAliased(int, std::string const&, google::protobuf::io::CodedOutputStream*)’未定义的引用
Cura.pb.cc:(.text+0xb000)：对‘google::protobuf::internal::WireFormatLite::WriteBytesMaybeAliased(int, std::string const&, google::protobuf::io::CodedOutputStream*)’未定义的引用
Cura.pb.cc:(.text+0xb041)：对‘google::protobuf::internal::WireFormatLite::WriteBytesMaybeAliased(int, std::string const&, google::protobuf::io::CodedOutputStream*)’未定义的引用
Cura.pb.cc:(.text+0xb082)：对‘google::protobuf::internal::WireFormatLite::WriteBytesMaybeAliased(int, std::string const&, google::protobuf::io::CodedOutputStream*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘cura::proto::GCodeLayer::MergePartialFromCodedStream(google::protobuf::io::CodedInputStream*)’中：
Cura.pb.cc:(.text+0xc429)：对‘google::protobuf::internal::WireFormatLite::ReadBytes(google::protobuf::io::CodedInputStream*, std::string*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘cura::proto::GCodeLayer::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const’中：
Cura.pb.cc:(.text+0xc546)：对‘google::protobuf::internal::WireFormatLite::WriteBytesMaybeAliased(int, std::string const&, google::protobuf::io::CodedOutputStream*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘cura::proto::Setting::MergePartialFromCodedStream(google::protobuf::io::CodedInputStream*)’中：
Cura.pb.cc:(.text+0x111a1)：对‘google::protobuf::internal::WireFormatLite::ReadBytes(google::protobuf::io::CodedInputStream*, std::string*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘cura::proto::Setting::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const’中：
Cura.pb.cc:(.text+0x11301)：对‘google::protobuf::internal::WireFormatLite::WriteStringMaybeAliased(int, std::string const&, google::protobuf::io::CodedOutputStream*)’未定义的引用
Cura.pb.cc:(.text+0x11342)：对‘google::protobuf::internal::WireFormatLite::WriteBytesMaybeAliased(int, std::string const&, google::protobuf::io::CodedOutputStream*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘cura::proto::SettingExtruder::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const’中：
Cura.pb.cc:(.text+0x124ab)：对‘google::protobuf::internal::WireFormatLite::WriteStringMaybeAliased(int, std::string const&, google::protobuf::io::CodedOutputStream*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘cura::proto::GCodePrefix::MergePartialFromCodedStream(google::protobuf::io::CodedInputStream*)’中：
Cura.pb.cc:(.text+0x133f1)：对‘google::protobuf::internal::WireFormatLite::ReadBytes(google::protobuf::io::CodedInputStream*, std::string*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘cura::proto::GCodePrefix::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const’中：
Cura.pb.cc:(.text+0x1350e)：对‘google::protobuf::internal::WireFormatLite::WriteBytesMaybeAliased(int, std::string const&, google::protobuf::io::CodedOutputStream*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘google::protobuf::internal::WireFormatLite::ReadString(google::protobuf::io::CodedInputStream*, std::string*)’中：
Cura.pb.cc:(.text._ZN6google8protobuf8internal14WireFormatLite10ReadStringEPNS0_2io16CodedInputStreamEPSs[_ZN6google8protobuf8internal14WireFormatLite10ReadStringEPNS0_2io16CodedInputStreamEPSs]+0x1f)：对‘google::protobuf::internal::WireFormatLite::ReadBytes(google::protobuf::io::CodedInputStream*, std::string*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘google::protobuf::internal::WireFormatLite::WriteStringToArray(int, std::string const&, unsigned char*)’中：
Cura.pb.cc:(.text._ZN6google8protobuf8internal14WireFormatLite18WriteStringToArrayEiRKSsPh[_ZN6google8protobuf8internal14WireFormatLite18WriteStringToArrayEiRKSsPh]+0x39)：对‘google::protobuf::io::CodedOutputStream::WriteStringWithSizeToArray(std::string const&, unsigned char*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o)：在函数‘google::protobuf::internal::WireFormatLite::WriteBytesToArray(int, std::string const&, unsigned char*)’中：
Cura.pb.cc:(.text._ZN6google8protobuf8internal14WireFormatLite17WriteBytesToArrayEiRKSsPh[_ZN6google8protobuf8internal14WireFormatLite17WriteBytesToArrayEiRKSsPh]+0x39)：对‘google::protobuf::io::CodedOutputStream::WriteStringWithSizeToArray(std::string const&, unsigned char*)’未定义的引用
lib_CuraEngine.a(Cura.pb.cc.o):(.rodata._ZTVN4cura5proto15SlicingFinishedE[_ZTVN4cura5proto15SlicingFinishedE]+0x20)：对‘google::protobuf::Message::GetTypeName() const’未定义的引用
[... cut ...]
/usr/local/lib/libArcus.so.1.1.0：对‘google::protobuf::compiler::Importer::Import(std::string const&)’未定义的引用
/usr/local/lib/libArcus.so.1.1.0：对‘google::protobuf::MessageLite::SerializeAsString() const’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/CuraEngine.dir/build.make:98: recipe for target 'CuraEngine' failed
make[2]: *** [CuraEngine] Error 1
CMakeFiles/Makefile2:142: recipe for target 'CMakeFiles/CuraEngine.dir/all' failed
make[1]: *** [CMakeFiles/CuraEngine.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

LipuFei, 2018-01-03:
@winyluo 你好，我们建议用静态链接来编译libArcus。你可以用`-DBUILD_STATIC=ON`来静态链接libArcus。

Ghostkeeper, 2018-01-22:
I don't know what's been said, but it seems that Protobuf couldn't be linked. This usually means that Protobuf was not compiled with the same compiler or that the header files don't match the object files.

Did static linking work for you?

BagelOrb, 2018-03-06:
Closing due to inactivity.

closed 2018-03-06 by BagelOrb