Ultimaker/CuraEngine #678: Fails to run on Arch Linux (Antergos) after 3.1.0-2 update

scar45, 2018-01-02:
Recently an update for CuraEngine was pushed 1/1/2018, but unfortunately this an error prevents Cura from starting. I believe it is related to `arcus`, which was also updated recently. I saw this in my updater logs:

`Warning: arcus: local (3.1.0-1) is newer than community (3.0.3-2)`
 
and trying to run cura afterwards leads to:

```
$> cura
Error in sys.excepthook:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3.6/site-packages/cura/CrashHandler.py", line 20, in <module>
    from UM.Application import Application
  File "/usr/lib/python3.6/site-packages/UM/Application.py", line 15, in <module>
    from UM.Settings.ContainerRegistry import ContainerRegistry
  File "/usr/lib/python3.6/site-packages/UM/Settings/ContainerRegistry.py", line 36, in <module>
    import UM.Qt.QtApplication
  File "/usr/lib/python3.6/site-packages/UM/Qt/QtApplication.py", line 15, in <module>
    from UM.FileHandler.ReadFileJob import ReadFileJob
  File "/usr/lib/python3.6/site-packages/UM/FileHandler/ReadFileJob.py", line 5, in <module>
    from UM.Application import Application
ImportError: cannot import name 'Application'

Original exception was:
Traceback (most recent call last):
  File "/usr/bin/cura", line 54, in <module>
    import Arcus #@UnusedImport
ImportError: libprotobuf.so.14: cannot open shared object file: No such file or directory
```

My `protobuf` version is `3.5.0.1-1`. This started happening after the last update to `CuraEngine (3.1.0-1 -> 3.1.0-2)`. 

I hope the above helps, but let me know if there's any additional info I can provide. Thanks!

awhiemstra, 2018-01-03:
It looks like your Arcus is compiled against an older version of protobuf and protobuf decided to introduce a break in ABI. So Arcus tries to find an older version of protobuf that no longer exists on your system. You will need to recompile Arcus (or prod the Arch maintainer to recompile) with the newer Protobuf.

LipuFei, 2018-01-03:
Actually I have `protobuf 3.5.1` and `libArcus` still links to it.

awhiemstra, 2018-01-03:
Yes, I am not disputing Arcus works with Protobuf 3.5. ;) My point is that the developers of protobuf changed the soversion to .15 or higher recently which breaks the packages on Arch. They need to be recompiled with the newer Protobuf so they properly link to the newer soversion. Since you are on Windows, in your case the protobuf library probably does not even have a soversion.

scar45, 2018-01-03:
Thanks for the detailed explanation @awhiemstra -- I understand what's going on now, and I will try to ping the Arch package maintainer for a rebuild against `libprotobuf.so.15`. Thanks again for your guidance!

scar45, 2018-01-03:
After getting in touch with the maintainer, he (rightfully so) claimed that there was no bug, and it was an issue with my machine. I can confirm this, as I removed and then reinstalled `cura`, `curaengine`, `arcus`, etc., and everything works fine. 

I am always learning, but what I am unsure of is how `arcus` somehow got updated to **3.1.0-1**, but after removing everything above, then installing again, `arcus` remained at **3.0.3-2** which works fine. Perhaps it's something related to Antergos and different repos?

I would love to hear more, to get a better understanding, but in the end, I apologize for opening up a bug here when I probably shouldn't have.

closed 2018-01-06 by scar45