Ultimaker/CuraEngine #679: Redundant travels causes lack of pressure in nozzle

cangelis, 2018-01-04:
Please download the [example project](https://www.dropbox.com/s/ewtv0log8td07sv/unnecessary%20moves.3mf?dl=0)

For example please go to layer `43`, and start moving the nozzle, you will see that even though there is only one wall in that layer, after printing first wall, nozzle goes to start point and travels without extruding as if there is second wall. This causes lack of pressure in the next layer because the material was oozed on the previous layer while traveling without extruding.

smartavionics, 2018-01-04:
I think it is printing 2 walls but the second wall is very thin (you have compensate wall overlaps enabled).

I recommend that you enable the wall order optimization (experimental section) so that the two wall lines start at the same place which will get rid of a lot of retracts/travels.

[image]

cangelis, 2018-01-04:
Thank you, I also tried disabling compensate wall overlaps, it still generates 2 walls and these 2 walls are overlapped, the resulting surface quality is not good as expected.

Does it suppose to generate only 1 wall? Because the wall thickness of the model is not wide enough to fit two walls.

smartavionics, 2018-01-04:
> Does it suppose to generate only 1 wall? Because the wall thickness of the model is not wide enough to fit two walls.

No, but one possible workaround is to use spiralize mode and then it will only print 1 wall. You may need to make the model solid for the spiralize to work right.

jackha, 2018-01-08:
If you want to use the current model, you cannot make the wall a single layer for sure. However you can fiddle around with:
- Horizontal expansion (can also be negative): this will make the model thinner
- Line widths

Ghostkeeper, 2018-01-23:
Do you happen to have an Outer Wall Inset set to something else than 0?

cangelis, 2018-01-29:
Thanks for your suggestions, horizontal expansion, increasing/decreasing line width, resizing and  sprialize options work for my case.

If you have this issue in some parts of the model rather than the whole like my example, it is not easy to determine, and if you have a big model, printing it with different parameters is a cumbersome, waste of time and material. 

I really like slic3r's approach for this problem that is just ignoring the overflowing walls. if your line width 0.4mm and the wall is 0.9mm it just prints 2 walls rather than printing 3 overflowing walls or "2 normal"+"1 empty wall".

cangelis, 2018-02-07:
Because the issue is unpredictable, most of the time, reprinting is required and parameters always need to be tweaked. Some example prints from the same issue.

[image]
[image]

Ghostkeeper, 2018-03-02:
I have that same book, Computational Geometry :smile:

Ghostkeeper, 2018-03-02:
There was a recent proposal that would say that if the wall overlap compensation reduces the wall to less than 30% or so it would just leave out the whole line.