Ultimaker/CuraEngine #682: Extrusion amount does not match line dimensions

Beherith, 2018-01-10:
Dear CuraEngine Devs,

I have been trying to use a firmware with linear advance, despite the advice online otherwise to not use it with Cura. I have found that Cura significantly over extrudes, and doesnt obey they exact [line width x layer height x line length] to extruded volume parameters set in it. 

For example, a print with 0.2 layer height, 0.4 line widths, e.g. 
```
;TYPE:WALL-INNER
G1 F1800 X131.224 Y109.6 E48.52414
G1 X131.224 Y110.4 E48.55075
G1 X88.776 Y110.4 E49.96258
G1 X88.776 Y109.6 E49.98919
G0 F9000 X88.376 Y109.2
M205 X5 Y5
;TYPE:WALL-OUTER
G1 F1200 X131.624 Y109.2 E51.42762
G1 X131.624 Y110.8 E51.48084
G1 X88.376 Y110.8 E52.91927
```

Take the difference between the last two lines, 43.248 mm line length, at 0.4 line width and 0.2 layer height that is a volume of 3.46mm3, with 1.438*(1.75*1.75) = 3.95mm3 of filament commanded to be extruded.

I have meticulously run through every possible setting (overlap compensations, flow rate) that influences flow rate, and tried to disable everything, yet the I have still found that Cura overextrudes by 15%!

Is there some setting that I am missing that ascertains that Cura produces the exact extruded amounts?

smartavionics, 2018-01-10:
>Take the difference between the last two lines, 43.248 mm line length, at 0.4 line width and 0.2 layer height that is a volume of 3.46mm3, with 1.438*(1.75*1.75) = 3.95mm3 of filament commanded to be extruded.

That should be 1.438 * (0.85 * 0.85 * PI) which comes to around 3.26 and that's actually less than the 3.46 mm^3 that the line should use so, if anything, it's underextruding, not overextruding.

Beherith, 2018-01-10:
Wow, I am completely retarded. I apologize for raising this issue needlessly, thank you very much for your help! Feel free to close as [user error].

closed 2018-01-10 by Beherith