Ultimaker/CuraEngine #683: Request: continuous toolpathing of shells

jaibooot, 2018-01-10:
Right now, The outer shells of features are discrete. The extruder typically retracts and unretracts when moving between shells. I would propose that it would be nice to have a continuous toolpathing of shells. Picture of what I mean is attached 

[image]

This way, total amount of retracts and unretracts will be reduced, hopefully increasing print quality. The outermost shell could end in an unretract, or travel across previously printed shells and begin printing the infill (dotted line)

smartavionics, 2018-01-10:
Have you tried using the wall print order optimization option that is available in recent Curas? Using that it will print the shells you show above left without any retracts. So if you have a part with multiple holes, it will print all the walls for each hole before moving on to the next hole.

jaibooot, 2018-01-15:
Thanks for the suggestion! That seems to help quite a bit. How about the end of the outermost wall? Is it possible to travel into the infill instead of retracting on the outer most wall and leaving a blob?

smartavionics, 2018-01-15:
Actually, the optimization I mention above does do that, it moves the nozzle away from the outer wall. It doesn't go very far, just about a nozzle width. You can see the movement if you zoom into the layer view. In this example, it travels from the outer wall of the hole (red) to the inner wall of the part's perimeter (green), prints the inner wall, then moves to the outer wall (red) then moves inboard again so it's approximately over the inner wall and then scoots off to do the infill.

[image]

jaibooot, 2018-01-15:
Just did some testing of this, and it definitely seems to work nicely! Thanks!

smartavionics, 2018-01-16:
Great, if you're happy, please close this issue. Cheers!

closed 2018-01-16 by jaibooot