Ultimaker/CuraEngine #686: using cura engine for print time estimation?

mark-hahn, 2018-01-21:
Is the Cura print time estimation done in the cura engine?  If so how hard would it be to make a command-line option to just read a gcode file and print the time estimate?  I would add this to octoprint since the current estimation in octoprint sucks.

Ghostkeeper, 2018-02-08:
The print time estimation is done in CuraEngine, yes. The way it works is to generate all the g-code first and then record how long each path takes to print, taking the relative acceleration from the previous moves into account and everything. So basically you'd need to run an entire slice to get the print time estimation from CuraEngine. And then it will be the estimate for that slice.

CuraEngine can't currently read g-code at all, only write it. That would need to be implemented. But I don't think that is within the scope of what CuraEngine should be doing.

mark-hahn, 2018-02-08:
>  But I don't think that is within the scope of what CuraEngine should be doing.

Probably true, but creating a spin-off utility for estimating print time would be very useful in a lot of environments, not just octoprint.  I wish I had free time.

closed 2018-03-02 by Ghostkeeper

Raul-Create, 2021-11-16:
Hey! Any advice on this point? I've run into this problem myself + there's bizarre differences in the time to print from the command line CuraEngine and the Win10 desktop app. I've tried to play around with transform matrices, but no luck. Any ideas?

Ghostkeeper, 2021-11-24:
Cura front-end calls upon CuraEngine under the covers to create that time estimate. The estimates should be identical.

However in Cura's front-end there are a bunch of profile layers applied in between, changing settings. It's pretty difficult and quite a lot of work to make all of the settings the same when you call CuraEngine via command line. I suspect you'd have different g-code as well then.