Ultimaker/CuraEngine #687: Generating objects of different quality

RNDAlex, 2018-01-23:
I was compile and run CuraEngine master branch in April 2017. 
But recently(December), I downloaded new master branch and did the same thing. The model generated by new engine was very different from the old one, yes, I used same options.

Like this:
`E:\\xxxc\\Cura\\CuraEngine.exe slice -m8 -j E:\\xxxc\\work\\fdmprinter.def.json -s retraction_enable="true" -s retraction_retract_speed="10" -s retraction_amount="2.5" -s z_seam_type="random" -s material_bed_temp_wait="true" -s machine_heated_bed="true" -s center_object="true" -s adhesion_type="skirt" -s top_thickness="0.8" -s top_layers="4" -s bottom_thickness="0.6" -s bottom_layers="3" -s support_mesh="false" -s infill_pattern="lines" -s retraction_extrusion_window="2" -s retraction_min_travel="0.8" -s machine_width="130" -s machine_depth="130" -s machine_height="130" -s mesh_rotation_matrix="[[0.3, 0, 0], [0, 0.3, 0], [0, 0, 0.3]]" -s layer_height="0.2" -s wall_thickness="0.8" -s infill_sparse_density="20" -s speed_wall="60" -s support_enable="true" -s machine_nozzle_size="0.4" -s material_diameter="1.75" -s material_print_temperature="200" -s material_bed_temperature="70" -p -l "E:\\xxxc\\Cura\\Feeder_-_Cap_731985.stl" -o "E:\\xxxc\\Cura\\Feeder_-_Cap_731985.gco"`

New model:
[image]

Old model:
[image]

New model is too fragile and it looks like a lack of material. The function about retraction in engine changed in new code? Thx for ur help

jackha, 2018-01-24:
Did you use a new fdmprinter file as well? I think a lot of settings has changed.

RNDAlex, 2018-01-24:
@jackha 
Of course I used new profile. But I checked retraction part in source code and wasnt find out any difference between new and old. BTW, I calculated the material  amount in gcode by hand and found which in new one is one-third of old model's

jackha, 2018-01-24:
Hmm sounds strange. 

Some ideas to check: Did you try Cura as well? Does it then give the desired results? You can then compare the settings sent to CuraEngine from Cura with the default settings.

RNDAlex, 2018-01-25:
@jackha 
My system is 32-bits and cannot install Cura with version 2.4 or higher.
I will show you guys gcodes I created before.
[Cura.zip](https://github.com/Ultimaker/CuraEngine/files/1662290/Cura.zip)

jackha, 2018-01-25:
@RNDAlex 
The new one indeed seems to under-extrude. However, the old one seems to over-extrude by a lot. At least "according" to our gcode parser / viewer. I've included screenshots so you can see what I mean.
Is it possible to send me the model? Then you can see how the g-code will be like.

New
[image]

Old
[image]

RNDAlex, 2018-01-25:
@jackha 
yea, I know what you said.
Thats why I thought the retraction part may have changed in new version.
[model.zip](https://github.com/Ultimaker/CuraEngine/files/1663359/model.zip)

RNDAlex, 2018-01-25:
Plus.
I ran the new engine in Linux and test the old engine in my Windows.
But... it should be the same.

jackha, 2018-01-25:
Here's a screen of a Custom FDM printer slicing the model at 30%. I turned the brim off and all the other settings are default. AFAICS it's very similar to your version. When I select different machines (UM2, 3), the results are somewhat different but still very similar. I'm pretty sure that these will print fine.

Maybe you can adjust the flow or so? If I set a flow of 250% then the result seems very similar to the result of the old slice.

[image]
[CFDMP_model.zip](https://github.com/Ultimaker/CuraEngine/files/1664218/CFDMP_model.zip)

RNDAlex, 2018-01-26:
@jackha 
Thx a lot! I think I found the answer.
I changed value of material_flow to 250 and printed well.
My material_diameter is 1.75. Maybe the value of material_flow is depend on material_diameter?
Or the implementation of handling parameter "e" in my machine is different from Ultimaker's
What still make me confused is why old engine work fine.

jackha, 2018-01-29:
Ah! You can also set the material_diameter parameter, that should solve it ^^. I think the engine takes 2.85 mm as a default.
Material_flow just multiplies whatever comes out by this amount. Normally a value of 100% should theoretically give the best results.

Ghostkeeper, 2018-02-14:
The engine doesn't take 2.85mm as default. In fact, the text `2.85` doesn't occur in CuraEngine's source code at all. It's not in the style of CuraEngine to hard-code that sort of thing.

jackha, 2018-02-15:
Yeah I meant fdmprinter

BagelOrb, 2018-03-06:
I suppose the problem is fixed now, seeing as there is no more activity on this issue.

closed 2018-03-06 by BagelOrb