Ultimaker/CuraEngine #689: Plans to clone makerbot "minfill"

flowchartsman, 2018-01-25:
Was wondering if there were any plans to research the [new "minfill" feature](https://www.makerbot.com/media-center/2017/03/20/introducing-makerbot-minfill) that Makerbot Print has. In short, it supposedly uses only the amount of infill required for any overhang, nothing more. Are there any plans to look into such a sparse infill, or are the existing ones such as concentric good enough?

ChrisTerBeke, 2018-01-25:
Cura already has gradual infill, which is not the same algorithm but fairly similar. At least it tries to solve the same problem (having too much infill where not needed).

Ghostkeeper, 2018-02-13:
If you only need the infill to get good overhang, I suggest turning on "Hollow Out Objects" and then turn on support as well, possibly even tree support. With tree support you'll get results that are very similar to Minfill.

BagelOrb, 2018-03-06:
We're not going to **copy** Makerbot.

Hollow Out Objects along with support does something which comes close.

My new Infill Support setting along with gradual infill does something close: https://github.com/Ultimaker/CuraEngine/pull/711

closed 2018-03-06 by BagelOrb