Ultimaker/CuraEngine #696: Make retraction from support to support configurable

mhcerri, 2018-01-29:
In some printers, limiting retraction from support to support causes oozing that leads to later under extrusion in the model. If that behavior is configurable, the issue can be avoided when necessary.

This behavior was introduced by the following the commit 14b1c5333aff ("fix: retraction was prevented for non-direct support to support moves which did avoid other parts (CURA-777)") and it's currently located at https://github.com/Ultimaker/CuraEngine/blob/master/src/LayerPlan.cpp#L318

MeerGiraffe40, 2018-02-06:
yes i think this is a nice idea. I am having the same problem with my actual print. 

But I think a setting called "retract on each travel move" would be perfekt. I use Cura 3.1.0 and printing ABS with a bowden set up. When the printer travels, he "prints" without feeding material, because the force of the bowden system builded up a pressure in the nozzle. 

But when the printer does retract everything works fine.

I would be really happy about some more retract settings.

mhcerri, 2018-02-18:
Any position on that?

BagelOrb, 2018-03-06:
I've made a Pull Request in my free time: https://github.com/Ultimaker/CuraEngine/pull/709

Prematurely marking this as fixed to clean up the amount of open issues in github.

closed 2018-03-06 by BagelOrb