Ultimaker/CuraEngine #697: Bad quality infill

wavexx, 2018-01-30:
Short description: Cura 3.1, printing with a Tevo Tarantula, 0.4 nozzle and PLA at 0.15mm height.
Shell and walls are perfect, layer adhesion is strong, but infill is trash irregardless of printing settings.

The root cause is the overlaying of filament in the same layer in crossed infill lines (such as grid pattern or cubic).

Now for the longer description:

Print a cube, 20x20x20mm, no shell, 30% infill with the grid pattern. With my setup, with 0.4mm nozzle and 0.15mm height, I'm hitting a sweet spot where the first few layers will come up fine, until cracks will form at intersections of the grid lines and build up vertically until the infill is essentially a stack of blobs having almost no horizontal interconnection. Speed, temperature, the filament itself as well the flow are almost irrelevant. Only the layer height has some influence. But I can design my own support grid and it will actually print *perfectly*.

The issue here is that for the same layer, each grid line is "flown" in full, without stopping the flow at intersections. This is causing a minimal amount of excess material to be squeezed at each crossing, until a little ridge is formed. After some layers, this ridge will accumulate and "wipe" the mouth of the extruder, causing a small void to appear just after the crossing. So in a crossing scenario like a "+", where the lines are filled top to bottom and left to right, the cracks will actually appear only on the lower/right sides of the crossing and propagate from there, reaching the further left/downward crossing.

When printing your own support grid this of course doesn't happen as each slice is never overprinted.

I don't want to give up this layer height, as it represents a sweet spot in terms of quality/speed for this machine. There are actually a few solutions to this.

Increasing the infill flow rate (via infill line width) _massively_ (150%) will flow enough material to fill all gaps produced at intersections. Unfortunately, this tends to produce quite a number of overextrusion problems nearby the walls that cannot be fixed even when coasting.

Decreasing speed, as previously mentioned, doesn't help. The ridge tends to form just slightly sooner than the crossing, but eventually still produces cracks.

Using an infill that doesn't overprint, such as "cross" (shouldn't this be "koch curve?"), works without problems. However, "cross" doesn't have the same structural characteristic.

I'm actually interested in "cubic", which unfortunately still exhibits the same problem. The advantage with cubic though is that the moving pattern will start the printing of the infill over slightly different spots over time. When a crossing is printed in reversed direction at each layer, no gaps are formed.

This seems to be key in printing good quality infill. Unfortunately, there's no way to force the infill to be forcibly printed in reversed direction at each layer. Setting the "z seam" property to "random" doesn't have any effect on the starting point of the infill (in fact, each layer is printed from exactly the same spot irregardless of this setting in my case!). I was hoping "optimize wall printing order", combined with "z seam" to "shortest" to cause a different starting point, but nothing changes. The infill is always printed from the closest point to the bottom/left of the bed.

wavexx, 2018-02-02:
Picture time! I took this in a hurry on a moving bed, so please forgive the bad quality.

[image]

This is a piece I've printed, 0.15mm height, and to exclude potential confounders, everything (including infill) is printed at 20mm/s and identical acceleration. I generally print outer walls a little faster still, but here I wanted to conserve as much as the inner infill as possible.

The cyan arrows indicate both the direction of travel and the first formation of the gap, just after the intersection. In some cases, it takes just 3/4 layers to form a gap. Since the geometry of this piece is variable, eventually the direction of the infill changes. The topmost direction is indicated by the red arrows. If there is a hole, it's generally healed by changing direction, and then is quickly reformed on the other side.

On the green circle I've highlighted where the little ridge is formed at almost any intersection due to the overprint.

If you look, overall the entire grid structure has this identical pattern across the piece, and it's extremely consistent in behavior.

Ghostkeeper, 2018-02-26:
> This is causing a minimal amount of excess material to be squeezed at each crossing, until a little ridge is formed. After some layers, this ridge will accumulate and "wipe" the mouth of the extruder, causing a small void to appear just after the crossing.

We've seen this too, but for my printer it really only becomes a problem when there's 3 lines crossing in the same place, such as for Triangle infill. It's less of a problem with Trihexagon or Grid.

I don't know a solution though, not with the same pattern. Switching to Zigzag or Lines already solves the problem, or even Cross/Cross3D. But those infill patterns have very different properties. Perhaps a Hexagon pattern will eventually solve this for you if we get to implement that.

Do you know what Cura could do better in order to fix this for Grid and Triangle infill?

wavexx, 2018-02-26:
On Mon, Feb 26 2018, Ghostkeeper wrote:
> Do you know what Cura could do better in order to fix this for Grid
> and Triangle infill?

I think I've described already what looks like a sensible solution.

Ensure that the infill direction is reversed at each layer. Each time
the direction is reversed, the previous layer is healed. If not
completely, almost completely in 2 passes.

Currently the infill is always started from the 0,0 corner. I could
think of a simple even-odd rule where in one case the infill is simply
started from the end in reverse.

Adjusting the flow rate at intersection would be interesting to do, but
I guess a bit more complicated to try.

Ghostkeeper, 2018-02-26:
Ah sorry. I had read a bit too fast.

The infill is started from the closest point where the previous extrusion ended up, not always the 0,0 corner. Changing the starting point to the other side will also not cause all lines to reverse. To reverse the lines you'd have to compute all lines first and store them separately, then iteratively reverse them. That could be considered but it's a bit of an ugly algorithm structure. In the new Connect Infill Lines algorithm of Cura 3.3 that can be done a bit more elegantly since that already produces a polyline that can be reversed easily.

Adjusting the flow rate at the intersection is unfeasible I'd say, since there is quite a delay between adjusting the extrusion rate and the effect of increasing the flow rate out of the nozzle. Even moreso if your printer has a bowden tube (that takes like at least a second). So that would be hard to predict.

wavexx, 2018-02-26:
On Mon, Feb 26 2018, Ghostkeeper wrote:
> The infill is started from the closest point where the previous
> extrusion ended up, not always the 0,0 corner.

I know, but the Z seam alignment is almost always forcing every layer to
start at the same position. The only noticeable difference in this
regard happens when using "random".

This is one of the reason "Cubic" has actually better results in my
case, since the pattern is constantly changing the intersection points,
until one of the intersections disappears, causing the filling direction
to start from another spot.

To try this fix though, you don't really need to enforce that every
single line is alternated. It would be fine to ensure it's a bit more
random.

I'm quite confident that if the direction is switched enough times (by
chance), the problem should be reduced.

I wanted to try and hack the engine myself with some horrible drudgery
just to see if it makes a difference, but couldn't find the time so far.

> Adjusting the flow rate at the intersection is unfeasible I'd say,
> since there is quite a delay between adjusting the extrusion rate and
> the effect of increasing the flow rate out of the nozzle. [...]

I suspected that. Dipping the nozzle _at_ the intersection to force some
squeezing could be more predictable, but still I see the layer direction
alternation as a way easier path.

BagelOrb, 2018-03-06:
> Each time the direction is reversed, the previous layer is healed. 

The total time a layer has to heal isn't changed. By alternating from where you start infill you give the first half of the previous layer more time to heal and the second part less time. I'm not sure how that would solve the problem.

wavexx, 2018-03-06:
On Tue, Mar 06 2018, Tim Kuipers wrote:
> The total time a layer has to heal isn't changed. By alternating from
> where you start infill you give the first half of the previous layer
> more time to heal and the second part less time. I'm not [...]

It's not about time, it's the travel direction that makes the
difference.

BagelOrb, 2018-03-06:
Ooohhh you mean that lines which are printed from front to back on the one layer should be printed from back to front on the next layer. 
```
>----->           <-----<
<-----<  becomes  >----->
>----->           <-----<
<-----<           >----->
```

Yeah that sounds like a good idea, but it's quite difficult to implement in CuraEngine. The direction of the lines is computed on the fly and the fact that lines seem to be in the same direction every layer is kind of a coincidence. (Actually it is caused by the fact that consecutive layers are generally almost the same shape.)

Ghostkeeper, 2018-05-30:
Our project manager removed this from our planning. The difficulty of implementing this means it's quite hard to fit in any planning. We won't get to spend time on this any time soon.