Ultimaker/CuraEngine #701: Feature Request/Question: Overlapp of perimeters in z-direction

pau33, 2018-02-09:
I print parts for my rc-models with cura.
They are often shells with 2 perimeters.

I wonder, if a overlap in z-direction of the two perimeter may get more strength.
Nozzle 0.4mm, layerheight 0.2mm.

Left is the way cura does it: the inner at the same z-direction as the outer.

Is it possible to generate an overlapp (right picture), so that the perimeters are shifted by 0.1mm in z-direction? To compensate one would need one outer layer with 0.1mm at the top an on 0.1mm inner layer at the bottom, or vice versa. So the layer boundaries would overlap.

Are my thoughts wright? I couldn't find such an issue yet. If there is already one, I apologize.

[image]

Ghostkeeper, 2018-03-05:
I don't think it will be stronger though. Here's a sketch of how this would look from the side.

[image]

It would probably be stronger if the walls are also put closer together.

However this would only have an effect in your case where the inside wall is directly touching the outside wall for the entirety of its length. If there's skin or infill in between, all bets are off because the skin wipe and infill wipe won't be wiping properly any more.

pau33, 2018-03-05:
Yes, when the walls are that far away, as you mentioned, than it is probably not much stronger. Probably you have to shift them together by about 0.05-0.1mm.
And you are also right, the inner wall (or all walls, if you use more then 2 perimeter) must be directly touching. That is often the case, if you print for example wing sections or parts of a fuselage.

BagelOrb, 2018-03-06:
While it is an interesting idea, it will take too much effort to implement this and there might be quite some defects caused by this.

Misalignment of the perimeters across two layers causes fundamental problems to this idea.

I'm closing this because Ultimaker will not spend time on this particular idea in at least the coming year, because there are more pressing issues.

closed 2018-03-06 by BagelOrb