Ultimaker/CuraEngine #702: Extraneous travel moves and extrusion width changes

jpwheless, 2018-02-10:
I'm using a Duet with pressure advance enabled, and I've noticed that some prints from Cura cause bad vibrations and extruder chatter.

The pressure advance feature treats the filament in a bowden tube like a spring in order to keep a constant extruder pressure vs print speed ratio.  It advances the filament a bit when extrusion is speeding up, and retracts it a bit when extrusion is slowing down, getting rid of underextrusion at the beginning of lines and overextrusion at the end of lines.

When the printer is moving through a series of line segments, like a curve, pressure advance doesn't come into play as long as the line widths and speeds are similar.  If the line width changes, or one of those segments is actually a travel move, then the extruder retracts and the print head might need to slow to obey extruder jerk/accel limits.

I took the gcode for one of these bad prints and loaded it back into Cura, and I believe it shows the moves that are causing my issues.

[image]

I took the gcode from Slic3r for the same mode, with similar settings, and loaded it into Cura to check.  It didn't show any extrusion width changes or small travel moves.

This picture shows the Cura gcode printed on the left, and a section of the Slic3r gcode on the right.  Both have the same pressure advance settings.  The surface finish on the left is clearly worse, and the main difference I can see are the travel moves and extrusion width changes.

[image]

Can thin walls and curves be treated more like the way Slic3r does?

smartavionics, 2018-02-11:
Just a thought, the wall overlap compensation is pretty broken and can give very uneven line widths so you may want to disable the overlap compensation if you are using it.

Also, I know that dc42 recently fixed something in the duet firmware that was related to pressure advance and uneven extrusion rates. So if you are not using the latest duet code, perhaps it's worth trying that.

Ghostkeeper, 2018-03-05:
These unnecessary travel moves seem to be the result of the last line sometimes fitting in between and sometimes not in between the walls. Adjusting the line width *slightly* either way (preferably slightly thinner though) would make the result a whole lot better.

cyounkins, 2019-02-15:
On a simple cube I see some seemingly unnecessary travel commands are because of a nonzero "Outer Wall Wipe Distance".

Ghostkeeper, 2019-02-18:
It's not unnecessary then, is it? :wink:

The-Distal, 2019-10-02:
(Doing this here, instead of opening a new issue). I believe that I may be experiencing a related issue to this one. I am using Cura 4.3.0 on an UM3e. I have noticed that I am consistently running into surface quality issues and extrusion issues, especially on thinner walled prints. 

I have documented my most recent struggles with this here: https://community.ultimaker.com/topic/29632-getting-better-cylinder-walls/

Basically, I noticed really poor walls on a recent print. I cleaned and calibrated my entire machine. The problems persisted. Finally, after trying may changes, I decided to swap in the gcode header into a file sliced by Slic3r, and instantly all the problems disappeared.  I then created a simple cylinder with 1mm walls to test with, and found that no setting in Cura seems to fully fix the problems.

I was finding extraneous travel moves in layers, and the extrusion was stuttering, when I checked the gcode Cura was generating. Turning off combing, gap fill, overlap compensation, zeroing the wipe, speed/temp setting tweaks, and tweaking the line width all eventually helped a little, but I'm still seeing poor surface quality. I've tried different nozzle/line width combinations, as I read that Cura may be better if the line/walls fit exactly, or if there is an even number of wall lines.

My last attempt was literally going through Cura turning as many features off as possible, to see what might be causing the issues... and I just have no clue. It almost looks like Cura is using a rougher approximation of the curve that partially drives the result. Even now, I've finally managed to get the gcode to look cleaner (can't find extrusion gaps or weird in layer moves), the outcome still looks much worse than the first Slic3r print. I changed the max resolution and deviation to 0.01mm to see if that would help... nope. I'm out of ideas. 

Photos and details can be found at the ultimaker forum post, but I am attaching the Cura project, stl, and the Slic3r gcode here for comparison. Note that these are both for a 0.8mm nozzle, but 0.40mm nozzle still exhibits the issues. Larger lines just make it easier to see in web photos. 

[Cura wall issue.zip](https://github.com/Ultimaker/CuraEngine/files/3681276/Cura.wall.issue.zip)

Ghostkeeper, 2019-10-03:
I recognise the problems in that forum thread but identify it differently. The problems with the g-code identified there will have an effect but I know of an effect that is probably much greater: Buffer underrun.

I believe that the printer firmware is not keeping up with the g-code. Could you run another test but in this one set the Maximum Resolution setting to 1mm and the Maximum Deviation to 0.15mm? See if that improves things.

The-Distal, 2019-10-03:
Ok, reprinted with the default 0.2mm profile, changed line width to 0.8mm (this is what the orange [Slic3r] comparison print was done with), and the only other changes are Max Resolution and Deviation changes as suggested, and shrank the brim to 2mm. 

No real difference. I'm redoing it with the 0.4mm nozzle, just for completeness, but that will take another hour to finish.

Apologies for the filament color. I ran out of the orange. The white is the same brand (filacube), and it has been very consistent for me. 

[image]

The-Distal, 2019-10-03:
And here's the 0.40mm nozzle version with the resolution/deviation adjustments. It's on the right.
[image]