Ultimaker/CuraEngine #703: Cannot build CuraEngine

martinberlin, 2018-02-10:
Maybe someone can give me a hint on what's the right way to do this:

I followed the Readme and installed as described:
Install Protobuf >= 3.0.0 (see below)
Install libArcus (see https://github.com/Ultimaker/libArcus)

But I've been most of this afternoon stuck in the latest part of compiling CuraEngine:

[100%] Linking CXX executable CuraEngine
/usr/bin/ld: lib_CuraEngine.a(commandSocket.cpp.o): undefined reference to symbol '_ZN6google8protobuf8internal10LogMessageC1ENS0_8LogLevelEPKci'
//usr/local/lib/libprotobuf.so.15: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/CuraEngine.dir/build.make:96: recipe for target 'CuraEngine' failed

What I'm doing wrong ?
libprotobuf.so.15 -> this is the version of protobuf that is installed after compiling 

$ protoc --version
libprotoc 3.5.1

Maye someone nows the solution ? My Ubuntu version is: 16.04 Xenial

Many thanks for reading this one!

alekseisasin, 2018-02-12:
I think the problem might be related to the version.
Please try to install libprotoc 3.0.0.
It might help you.

thopiekar, 2018-02-13:
Or since you have the PPA enabled, as far as I know, then just install p4otobuf from there 😉👌

alekseisasin, 2018-02-15:
@martinberlin , Did you try our tips.

wawanbreton, 2024-04-05:
Closing this discussion as the build method is really different now.