Ultimaker/CuraEngine #708: extruder end/start position does not work correctly in relative mode

sofino, 2018-02-26:
Setting "machine_extruder_start_pos_abs" to "false" does not result in a relative move of the extruder after its last position. 
The behavior only occurs when starting a new layer and immediately switching the extruder. 
Each layer has the last_planned_position of the print head, by default if the current layer just started, the last planned position is 0;0, and only then the actual position of the print head will be set to last_planned_position.
So, when a new layer has just started, last_planned_position is 0; 0, and the extruder will immediately switch before the actual position of the print head is set to last_planned_position. This leads to the fact that during the switching the extruder is homing using these parameters “machine_extruder_end_pos_x/y”, since we just started the current layer, the last planned position is 0;0, therefore the extruder moves from 0;0 instead of moving from its last position.

BagelOrb, 2018-03-06:
I don't think this is a problem for any of our printers, so Ultimaker won't put time in fixing it.

I find it a little hard to understand what you think the problem is. The issue template asks for what you expect to happen, but I guess the template is only set in place for the Cura repo.

The setting machine_extruder_start_pos_abs set to true means that when you start this extruder (after an extruder switch) the machine will move to the position machine_extruder_start_pos_x/y.
Setting it to false means that the print head *should* make a relative move from its current position with the vector given by machine_extruder_start_pos_x/y.

You seem to understand the problem very well. Perhaps you could make a Pull request to fix the problem yourself and have it incorporated in the next Cura version.

Ghostkeeper, 2018-03-29:
No in fact I think this is a duplicate of https://github.com/Ultimaker/Cura/issues/2433

I made a similar analysis by looking into the code. I find it impressive that you came to the same conclusion by looking at just the behaviour!

closed 2018-03-29 by Ghostkeeper