Ultimaker/CuraEngine #712: Error : could not found OpenMP (missing: OpenMP_FOUND)on Mac osX

cliegh, 2018-03-05:
I want to run curaEngine on mac at vscode. i follow readme.. after install protobuf and libarcus, git clone, $ mkdir build && cd build. and then i command cmake..
-- Building with Arcus
-- Compile with more optimization flags
-- Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES)
-- Found OpenMP_CXX: -fopenmp
-- Could NOT find OpenMP (missing: OpenMP_C_FOUND)
-- Configuring done
-- Generating done
-- Build files have been written to: /CuraEngine/build

actually i search about that issue so adding below code to CMakeLists.txt
```
set(CMAKE_C_COMPILER "/usr/local/Cellar/llvm/5.0.1/bin/clang")
set(CMAKE_CXX_COMPILER "/usr/local/Cellar/llvm/5.0.1/bin/clang++")
set(OPENMP_LIBRARIES "/usr/local/Cellar/llvm/5.0.1/lib")
set(OPENMP_INCLUDES "/usr/local/Cellar/llvm/5.0.1/include")

OPTION (USE_OpenMP "Use OpenMP to enamble <omp.h>" ON)

# Find OpenMP
if(APPLE AND USE_OpenMP)
    if(CMAKE_C_COMPILER_ID MATCHES "Clang")
        set(OpenMP_C "${CMAKE_C_COMPILER}")
        set(OpenMP_C_FLAGS "-fopenmp=libomp -Wno-unused-command-line-argument")
        set(OpenMP_C_LIB_NAMES "libomp" "libgomp" "libiomp5")
        set(OpenMP_libomp_LIBRARY ${OpenMP_C_LIB_NAMES})
        set(OpenMP_libgomp_LIBRARY ${OpenMP_C_LIB_NAMES})
        set(OpenMP_libiomp5_LIBRARY ${OpenMP_C_LIB_NAMES})
    endif()
    if(CMAKE_CXX_COMPILER_ID MATCHES "Clang")
      set(OpenMP_CXX "${CMAKE_CXX_COMPILER}")
      set(OpenMP_CXX_FLAGS "-fopenmp=libomp -Wno-unused-command-line-argument")
      set(OpenMP_CXX_LIB_NAMES "libomp" "libgomp" "libiomp5")
      set(OpenMP_libomp_LIBRARY ${OpenMP_CXX_LIB_NAMES})
      set(OpenMP_libgomp_LIBRARY ${OpenMP_CXX_LIB_NAMES})
      set(OpenMP_libiomp5_LIBRARY ${OpenMP_CXX_LIB_NAMES})
    endif()
endif()

if(USE_OpenMP)
  find_package(OpenMP REQUIRED)
endif(USE_OpenMP)

if (OPENMP_FOUND)
    include_directories("${OPENMP_INCLUDES}")
    link_directories("${OPENMP_LIBRARIES}")
    set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} ${OpenMP_C_FLAGS}")
    set (CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${OpenMP_CXX_FLAGS}")
    # set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} ${OpenMP_EXE_LINKER_FLAGS}")
endif(OPENMP_FOUND)
```
then.. get that results
-- Building with Arcus
-- Compile with more optimization flags
-- Found OpenMP_C: -fopenmp=libomp -Wno-unused-command-line-argument
-- Found OpenMP_C: -fopenmp=libomp -Wno-unused-command-line-argument
-- Found OpenMP_CXX: -fopenmp
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/soonwon/soonwon/hci/CuraEngine/CuraEngine/build

i think i solve that issue so i try **make** command but ```warning: use of this statement in a constexpr function is a C++14 extension [-Wc++14-extensions]``` are appeared. also ```error: 'va_start' cannot be used in a captured statement``` and ```warning: macro expansion producing 'defined' has undefined behavior [-Wexpansion-to-defined]``` and so on..

I search about this issue. but i cannot find a similar case. so please help me how could i run curaEngine on mac osX. thx

BagelOrb, 2018-03-06:
I don't know much about building CuraEngine, but I thought it was possible to build it without OpenMP.

If you can't get it to build without it, try installing it ;)

Ghostkeeper, 2018-03-29:
You can prevent usage of OpenMP by disabling the CMake variable `ENABLE_OPENMP`.

We currently disable that for MacOS because the version of MacOS that we use for the official Cura builds isn't supported by OpenMP. I understood that for more modern versions of MacOS it is supported, but we use an old MacBook so that users with old Mac computers will be able to run Cura as well.

I think that pretty soon it's time to upgrade that MacBook to a slightly newer MacBook which does support OpenMP so that slicing will be much faster for MacOS users.

WinstonMao, 2018-12-14:
I have the same problem when building CuraEngine in macos,but when I use gcc/g++ as  Compiler to build CuraEngine works well

wawanbreton, 2024-04-05:
Closing as the build method is really different now.

closed 2024-04-05 by wawanbreton