Ultimaker/CuraEngine #713: Build CuraEngine into Visual Studio

asegurpe, 2018-03-05:
Hi,

I contact you because I'm trying to build CuraEngine in Visual Studio, but I cannot.

I view that is necessary compile libArcus and previously protobuf, but also I have got problems.

Thanks for your help!

ianpaschal, 2018-03-05:
>I have got problems.

We all do...

Can you explain more about what doesn't seem to work? What kind of errors are you getting.

Ghostkeeper, 2018-03-29:
We normally build CuraEngine using MinGW when building for Windows. It's the only compiler we really support for Windows. We got a PR once to fix some things that were not supported by Visual Studio, but we won't fix these ourselves.

Unless you got a very clear error message for us and we can spend a few minutes to fix it for you with very little risk. Then it's fine of course. But then you might as well make a PR.

The normal process for building the entire Cura software with front-end is to compile libArcus and Protobuf with MinGW and with Visual Studio (so compile it twice). Then link the libArcus version to CuraEngine and link the VS version to Python.

closed 2018-03-29 by Ghostkeeper