Ultimaker/CuraEngine #716: [3.2.1][2.7]Print job "larger than build volume" issue ultimaker3 extended

recarens, 2018-03-09:
With curaEngine 3.2.1 I created a gcode correctly for Ultimaker3 extended with the default parameters of de ultimaker3_extended.def.json and the printer shows me: "This print job is larger than the printer's build volume." and if i load the same gcode from a Cura2.7 it loads correctly and i can print it.

For example, i can't print this object. It's build in CuraEngine 3.2 

[image]

Loading it from the usb :
[image]

alekseisasin, 2018-03-09:
Hi @recarens , We have tried to reproduce the issue and we don't have this bug. Are you sure that you uses latest version of Cura, CuraEngine and printer definition files. Maybe you can share your project here and we will test it.

recarens, 2018-03-09:
I found the problem 

in G-code created by curaEngine:

;PRINT.SIZE.MIN.X:2.14748e+06
;PRINT.SIZE.MIN.Y:2.14748e+06
;PRINT.SIZE.MIN.Z:2.14748e+06
;PRINT.SIZE.MAX.X:-2.14748e+06
;PRINT.SIZE.MAX.Y:-2.14748e+06
;PRINT.SIZE.MAX.Z:-2.14748e+06

I edit this by:

;PRINT.SIZE.MIN.X:9
;PRINT.SIZE.MIN.Y:6
;PRINT.SIZE.MIN.Z:0.27
;PRINT.SIZE.MAX.X:127.396
;PRINT.SIZE.MAX.Y:119.238
;PRINT.SIZE.MAX.Z:2.07

IT WORKS

recarens, 2018-03-09:
But i have to automatize the task , i need to find the parameter.

alekseisasin, 2018-03-09:
What do you mean automatize, the output which you showed is wrong, if CuraEngine produces it then something goes wrong. Do you use CuraEngine for slicing without Cura UI( you call slicing via console) ?

recarens, 2018-03-09:
yes, im using comand line. 
Example:
cura3_2/CuraEngine slice -v -j cura3_2/Resources/resources/definitions/ultimaker3_extended.def.json -l object.stl -o object.gcode

Ghostkeeper, 2018-03-09:
This was fixed here: https://github.com/Ultimaker/CuraEngine/commit/ae543d96a441dcc4cd54ef9daa57ca5fdffd52e5#diff-023d354b12b61c267cf78a465c76f3fc

With Cura 3.3 it should put a 1cm^3 volume as fallback, which gets accepted by the printer.

The problem is that the volume is unknown at the point that this header needs to be output (before the g-code is generated).

Ghostkeeper, 2018-03-09:
Duplicate of https://github.com/Ultimaker/CuraEngine/issues/660

closed 2018-03-09 by Ghostkeeper