Ultimaker/CuraEngine #717: the amount of displayed layers is wrong

UltiCa, 2018-03-09:
think, I have located a bug in Cura 3.2.1

as you can see in the Pictures, the model has only two layers. The original size is 0.5 mm height. First layer 0.3mm and the next 0.2mm =0.5 mm. In the layerview, there are 3 layers displayed as you can see. 
[image]
[image]
[image]

diegopradogesto, 2018-03-16:
Yes, this is a known issue.
However if you see the horizontal slider, the last layer is empty, so actually there are just two layers. The third layer is because at the end we receive a movement in Z and it counts as a new layer in layer view, but nothing is printed.

Ghostkeeper, 2018-04-16:
This is fixed in version 3.3.

closed 2018-04-16 by Ghostkeeper