Ultimaker/CuraEngine #720: How do I change curaengine settings to avoid unnecessary path in some special models

RNDAlex, 2018-03-12:
Like this model:
[image]

Now I slice it with CuraEngine:
[image]

But what I expected:
[image]

Thx for ur help!

diegopradogesto, 2018-03-16:
Have you tried the Combing Mode (setting called retraction_combing) and set it to "all"?

Let me know if it works as you expect.

RNDAlex, 2018-03-20:
@diegopradogesto 
no help. no matter I set it "all" or "off"

smartavionics, 2018-03-20:
Could you please make the model and the gcode that you don't like available, please. It's not really possible to understand what is happening from just the images. Thanks.

RNDAlex, 2018-03-21:
In fact, the model in the third picture is another model, but that's what I want my model sliced to.
I tried other software which with CuraEngine like repetier-host. The result is not perfect but better than raw engine did.(number of lines between outline tower much less)
[image]

RNDAlex, 2018-03-21:
Something wrong with the process that uploading the files to Github. I will try again later.

Ghostkeeper, 2018-04-17:
Maybe you need to set your travel avoid distance a bit lower.

To upload to Github, put the file(s) in a .zip. Github accepts zips.