Ultimaker/CuraEngine #723: [a05c84e] layer_nr should be size_t to match slice_layer_count in src/slicer.cpp:818

onitake, 2018-03-13:
(Sorry for not using the template, but this is a source bug, not an application bug)

In src/slicer.cpp#L818, the for loop uses a loop variable of type unsigned int, but the value it's compared against is of type size_t. This is not an issue on most platforms, but can be problematic when size_t is 64 bit and int is 32 bit, for example.
The type of the loop variable should be changed to size_t.

Thanks.

onitake, 2018-03-13:
There are further instances of unsigned int further below - they should probably all be size_t or vector<>::size_type.

See: http://en.cppreference.com/w/cpp/container/vector/size and http://en.cppreference.com/w/cpp/container/vector -> size_type

diegopradogesto, 2018-03-16:
Yes, we are changing those as soon as when we found them in the code. We have planned a major refactor in CuraEngine so probably then it will be a good moment for this.

BagelOrb, 2018-03-20:
I don't see the problem. Why would you want to be more explicit, rather than letting the compiler choose an appropriate type for each system?

2 billion (32 bit max) is high enough for a vertex count or whatever variables occur in slicer.cpp.

onitake, 2018-03-21:
There are multiple problems here:
- It's not just the compiler that chooses the type, it's also the platform and ABI
- Intermixing types can lead to behaviour inconsistencies between different platforms
- Index variables should never have intermixed types (including signed/unsigned), because that's one of the roads to a buffer overflow vulnerability, _especially_ in user-controlled or externally-supplied data
- Never assume anything about the nature of user-supplied data (not even "it can never have more than 2³² records")
- Better never introduce type confusion in the first place, but at least fix it when you find it

That's precisely the reason why modern strong-typed languages always define a precise contract on the range of their types and never have something like C's int.

BagelOrb, 2018-03-21:
I still have a lot to learn I guess.

* The compiler chooses the type of int based on your platform. It optimizes the int type for your platform.
* Intermixing types can only cause inconsistencies if the smallest type is too small to contain the data it's supposed to. So the problem is then that the type is too small for the data.
* Why should I care about buffer overflow vulnerability? Is that an important thing for a slicer program?
* I have to assume the size of the user-supplied data. I either assume it will fit in 32 bit or it will fit in 64 bit. I'm not gonna use 128 bit int or 256 bit int either. I have to choose a type which is appropriate for the task and I think 32 bit will do here.
* What do you mean by "type confusion"?

onitake, 2018-03-21:
- A C or C++ compiler never optimises on types. It has a fixed set of types that are represented by a particular, platform-specific type when code is compiled for that platform. The only time you would actually see type optimisation is when using constant values or when doing arithmetic where a shorter or longer type is more efficient and does not influence the result. This very different from dynamically typed languages like Python or JavaScript, where a sensible type is chosen for each operation and representation, and type conversion is implicit and can change the result.
- This is exactly the case. But since you can not know in advance what type is chosen by the compiler when you don't use well-defined types like size_t or uint32_t,  you can not know where inconsistencies will arise.
- It is very much so. Consider that CuraEngine will be used by random people with random data, possibly downloaded from the internet, it may be very, very easy to trick users into exposing their computer to vulnerabilities. Also consider the case where CuraEngine is operated on a server where random people can upload data.
- The best thing to do is to rely on the "best" type for a certain operation. In the case of binary data, this will depend on the accuracy of the data (for example a int32_t or a uint8_t). In the case where memory is indexed, it is best to rely on a well-defined type that can not lead to over- or underflow, such as size_t. If you use smart types that do range checking, the solution is a bit easier, but it will always be a good idea to use a type that matches what the smart type expects. Or, use the recommended addressing method of the type: In case of a std::vector, this would be an iterator. That way, you also won't impose artificial limitations on handled data.
- It basically means assuming supplied input data is of a certain type (like an unsigned 32 bit integer), while the data actually represents something different. Sometimes, this can lead to crashes or errors with invalid data, but in the worst case you will never suspect anything until somebody exploits it for malicious purposes. See [here](https://cloudblogs.microsoft.com/microsoftsecure/2015/06/17/understanding-type-confusion-vulnerabilities-cve-2015-0336/) or [here](https://media.ccc.de/v/34c3-8848-type_confusion_discovery_abuse_and_protection) for somewhat technical discussions of the topic.

Now, I'm not saying one has to panic and spend sleepless nights fixing code. But it's always a good idea to review one's code from time to time and address possible issues before they become a problem. CuraEngine is a very valuable piece of software and receives a lot of attention. It makes only sense to take a close look at its internals. The good thing is that it's also open source, so many more eyes can do that. :)

BagelOrb, 2018-03-21:
Hmm. I did some looking and it seems that for the reasons I wanted to use int I should actually have used things like uint_fast16_t.

I just wrote a loop like this:
```
for (uint_fast8_t child_idx_offset = 0; child_idx_offset < 4; child_idx_offset++)
```

Thinking of keeping this style of using numeric types. What do you think?

onitake, 2018-03-22:
I'd generally use size_t for a loop index variable, since that will usually map to an efficient index register and give you enough headroom for avoiding programming mistakes. If you know in advance that the range is limited to 0..254 (not 255, if you implement the loop this way), then yes, something like uint8_t may make sense.

I occasionally do some microcontroller programming, and in that context it matters what type you choose. But for 32 bit or 64 bit computers, size_t is usually the best choice.

But I guess it doesn't make much of a difference in this case, as a smart compiler will try to unroll loops with a constant range anyway...

onitake, 2018-03-22:
The most dangerous case for C/C++ is probably intermixed signed and unsigned types, particularly where the signed value can be supplied externally.

Consider this example:
```c++
struct context_t {
    int *data;
    int size;
};
void update_value(context_t *ctx, int index, int newvalue) {
    if (index < ctx->size) {
        ctx->data[index] = newvalue;
    }
}
```
If someone manages to pass a negative value to this function, the range check would not fail, because of the signed comparison. The index register would happily underflow and generate a write access to memory _before_ data. This could have disastrous results.
The solution would be to check for index >= 0, or to use an unsigned type (like size_t) in the first place. It's still possible to pass a very large value, but the range check would easily detect that.

BagelOrb, 2018-03-22:
I'd like to get some other opinions in, but nobody else is joining the
discussion.

@ghostkeeper what do you think: int vs size_t vs int32_t vs int_fast32_t

I get that intermixing signed and unsigned is problematic. However choosing
whether a variable is signed is not always trivial.

Example
```
for (unsigned int array_index = size -1; array_index >= 0; array_index++)
```

You either have to make it a signed int or change the condition to
`static_cast<int.>(array_index) >= 0` OR to `array_index !=
static_cast<unsigned integer>(-1)`

onitake, 2018-03-23:
I'd generally avoid reverse loops (assuming you meant array_index-- in your example) if possible, but there are actually two safe approaches to this problem:

```c
for (size_t array_index = size - 1; array_index-- > 0;)
```
This looks confusing and relies on the fact the array_index-- post-decrements, with the comparison happening after the decrement. But it's the generally accepted solution for C code.

In C++ you could use a reverse iterator, which is much more idiomatic and safer:
```c++
std::vector<xxx> blob;
for (auto it = blob.crbegin(); it != blob.crend(); it++)
```
Note that incrementing the reverse iterator _decrements_ the index here.

BagelOrb, 2018-03-23:
Oh that's a nifty trick. Code is less readable in that first example, though.

Problem with iterators is that you don't have the index anymore. For example the following trick is not possible any more:

```
float computeSomething(const std::vector<Item>& data)
{
   std::vector<ItemStatistics> data_decoration(data.size());
   for (int index = 0 ...)
        data_decoration[index] = computeSingleStatistic(data[index]);
}
```

Ghostkeeper, 2018-04-12:
> The compiler chooses the type of int based on your platform. It optimizes the int type for your platform.

The problem with this is also that the compiler runs on our build system which is not the same as the platform that the user is running on.

> @ghostkeeper what do you think: int vs size_t vs int32_t vs int_fast32_t

Generally I've been using size_t for iterating over arrays. I didn't choose this specifically for buffer overflows because in general the computer will run out of memory long before the limit of unsigned int is reached (you're going to have trouble storing 4.2 billion layers in RAM). I chose it more because of semantics: `size_t` is easy to recognise as being the index to some array. I'm also a strong proponent of using `coord_t` as a type where it concerns g-code coordinates. In a personal project I've also made `area_t`, just for semantics.

I've been using the boy scout rule for this though: Fix it when you touch the code, but leave the rest in.

For mixing signed and unsigned integers I get a compiler warning. Currently we don't have that in our code and if we do we'll notice pretty quickly.

I've rarely had a need to do a reverse iteration with an index, but I've been known to use something like this, like BagelOrb said:

    for (size_t i = arr.size() - 1; i != static_cast<size_t>(-1); i--)

The cast of `-1` to `size_t` causes an underflow. And when the `i` variable becomes `0` and then computes `i--` it causes the same underflow which then stops the loop. I think I like Onitake's safe approach better since it's safer if the type changes. Might use that from now on.

Ghostkeeper, 2018-04-12:
> Yes, we are changing those as soon as when we found them in the code. We have planned a major refactor in CuraEngine so probably then it will be a good moment for this.

I don't think a major refactor of the settings in CuraEngine is a good moment to change this. You're essentially doing two refactors at the same time then.

BagelOrb, 2018-04-12:
> The problem with this is also that the compiler runs on our build system which is not the same as the platform that the user is running on.

I don't think the compiler optimizes for your build system, but instead for the build _target_ platform.

I've done some reading and I am now convinced that using `int` is bad and I should feel bad. The size of int is undefined, except that it is at least 1 byte -_-

Using intN_t (like int32_t) doesn't really make sense in most contexts. Generally you want to ensure that you can _at least_ fit some data in a type.

However, I am not so much convinced of the one `size_t` fits all point of view.
I still feel more for thinking about what a reasonable size of some variable should be and then using `int_fastN_t`. 
I'm still undecided.

Side note: I've just read that `signed char` is not synonymous with `char`! This is madness!

onitake, 2018-06-20:
From the gcc manpage:
```
       -funsigned-char
           Let the type "char" be unsigned, like "unsigned char".

           Each kind of machine has a default for what "char" should be.  It is either like "unsigned char" by default or like "signed char" by default.

           Ideally, a portable program should always use "signed char" or "unsigned char" when it depends on the signedness of an object.  But many programs have been written to use plain "char" and expect it to be signed, or
           expect it to be unsigned, depending on the machines they were written for.  This option, and its inverse, let you make such a program work with the opposite default.

           The type "char" is always a distinct type from each of "signed char" or "unsigned char", even though its behavior is always just like one of those two.

       -fsigned-char
           Let the type "char" be signed, like "signed char".

           Note that this is equivalent to -fno-unsigned-char, which is the negative form of -funsigned-char.  Likewise, the option -fno-signed-char is equivalent to -funsigned-char.
```

My opinion is that size_t is the right type for anything that indexes an array, because size_t represents the type that is closest to the target architecture's native type for that purpose. It is also used by the C++ STL where appropriate.
For anything that interacts with data files, network protocols or binary APIs, a precise type like uint32_t should be chosen.
Other than that, you can probably get away with int, short or the like. But you should be aware that there might be an exotic system out there that does not behave the way you expect it to.

Other than, you should be careful not to confuse signed/unsigned, as that can lead to other problems...

bjude, 2018-08-20:
@BagelOrb, in reference to your post on March 23, your code snippet:
```
float computeSomething(const std::vector<Item>& data)
{
   std::vector<ItemStatistics> data_decoration(data.size());
   for (int index = 0 ...)
        data_decoration[index] = computeSingleStatistic(data[index]);
}
```

is probably better expressed as:
```
float computeSomething(const std::vector<Item>& data)
{
   std::vector<ItemStatistics> data_decoration(data.size());
   std::transform(data.begin(), data.end(), data_decoration.begin(), &computeSingleStatistic);
}
```

if the loop index is only needed for indexing into a couple of vectors like that then theres really no need for the index variable at all. This falls over if you have to iterate through more than 2 containers at a time will cover the majority of simple cases like this. If you really need the iteration index then size_t is generally the appropriate type for consistency with the STL. 
It's a shame that size_t is unsigned but thats a whole other kettle of fish

onitake, 2018-08-20:
@bjude Good suggestion. Use the STL!
If you really need a signed index variable, there is ssize_t, by the way...