Ultimaker/CuraEngine #733: Layer Height Parameter Not Changing Number of Layers

kennethshawfriedman, 2018-03-26:
I am feeding CuraEngine an STL file with the Ultimaker3 json config file, and then sending it the `-s layer_height='0.2'` parameter.

Regardless of what layer_height I give it, it produces GCode with the number of layers as if layer_height=0.1. Are there other settings I also have to change? Other parameters and settings seem to be working just fine.

BagelOrb, 2018-04-03:
You are probably supplying the setting at the wrong point in the command line call. Try setting that setting at the very start.

Reread the help and try to figure out how the order determines the slicing.

BagelOrb, 2018-04-03:
As in `CuraEngine help`

Ghostkeeper, 2018-04-26:
In particular, ensure that the `-s layer_height=0.2` occurs before any -E parameters, otherwise the layer height would only get set for one of the extruders and that isn't supported.

closed 2018-04-26 by Ghostkeeper