Ultimaker/CuraEngine #737: [3.3] new "continuous" infills should also apply to support structures

jaibooot, 2018-03-30:
**Application Version** : 3.3

The new "continuous" toolpaths are a huge benefit, especially for hard to print materials. The new infill toolpathing should also apply to support structures.

smartavionics, 2018-03-30:
That would be good as I often find that support is missing around the perimeter of the area being supported so by zig-zagifying the support lines, there's more chance the support will be their on the perimeter.

BagelOrb, 2018-04-03:
This should be really easy to implement. Support uses the same infill class as infill to generate it's structures..... It's just a matter of introducing the setting.

Ghostkeeper, 2018-04-26:
We didn't do this initially because the connected infill lines don't guarantee that the same pairs of lines get connected on the same layer. This produces support where the connecting lines are hanging in mid-air. You can see this effect in layer view in the infill when you slice a sphere and disable the visibility of the shell:
[image]
That's just not going to print well if there's not a wall around it to sort of support it.

We'll have the option to put a wall around support soon (thanks to BagelOrb). Then this might become feasible.