Ultimaker/CuraEngine #742: [3.3] When using interface in small z height support areas,  infill prints before skin

jaibooot, 2018-04-09:
**Application Version** - cura 3.3 beta, although I believe this has been a bug for a while

**Steps to Reproduce**
Upload stl file with a gap that will build a support structure that is <~2mm in z. Enable support interface extruder with E1 

**Actual Results**
The part does not print top/ bottom layers on top of the support structure. Instead, it prints multiple layers of infill before printing the top / bottom layers. 

[image]
**infill printing above support structure**
[image]
**top/ bottom layers that print a few layers above the support structures**

[UM3_curatest.gcode.gz](https://github.com/Ultimaker/CuraEngine/files/1890587/UM3_curatest.gcode.gz)
[UM3_curatest.gcode.gz](https://github.com/Ultimaker/CuraEngine/files/1890588/UM3_curatest.gcode.gz)

[cura 3.3 support gap test.zip](https://github.com/Ultimaker/CuraEngine/files/1890590/cura.3.3.support.gap.test.zip)

BagelOrb, 2018-04-17:
Disabled the setting Ignore Small Z Gaps.

BagelOrb, 2018-04-17:
I mean: please disable ...

jaibooot, 2018-04-17:
understood, thank you!

closed 2018-04-17 by jaibooot