Ultimaker/CuraEngine #743: Concentric method leaves gap in top/bottom layer

midgethoen, 2018-04-14:
**Application Version**
3.2.1

**Platform**
OSX

**Steps to Reproduce**
Slice stl.. 🥁tss 

**Actual Results**
Cura sometimes leaves a gap in the top/bottom layer with the concentric method.
[image]
[image]
[image]

**Expected results**
This gap should have been closed

**Additional Information**

sample stl and cura profile:
[Archief.zip](https://github.com/Ultimaker/CuraEngine/files/1910322/Archief.zip)

BagelOrb, 2018-04-17:
Do you have the setting Fill Small Gaps enabled?

midgethoen, 2018-04-17:
I did not even know it existed. I will try slicing again tonight and see if that helps.

thanks for the tip

midgethoen, 2018-04-18:
> Do you have the setting Fill Small Gaps enabled?

I could not find this setting. is this its exact name?

BagelOrb, 2018-04-18:
Probably not. Just put "gaps" in the setting search bar. I don't have a
computer ATM.

BagelOrb, 2018-04-18:
It's about the setting above that, but that one is already set to everywhere, so there's a bug.

genghisnico13, 2018-04-23:
Just in case, here is the first layer of a Benchy, cura V3.2.1 in Linux.
[image]
I tried changing the 3 mentioned gaps options with no difference in this layer. The printed part has those gaps as expected.
Fill gap between walls should do something similar to the thin segment between the 3 and the D but everywhere?

diegopradogesto, 2018-04-23:
I think those spaces are there because there is no space enough for another infill line. We don't have a feature to fill tiny gaps in infill, as far as I know. We have it just for the tiny space between walls.

BagelOrb, 2018-04-23:
`FffGcodeWriter::processSkinPart` calls `processPerimeterGaps`, so this is a bug which is either recently introduced or nobody ever noticed it was introduced.

BagelOrb, 2018-04-23:
@diegopradogesto This issue is not about infill, but about skin!

diegopradogesto, 2018-04-23:
Yep, sorry, skin was in my head but I wrote infill.
Devs. CURA-5276

AbeFM, 2018-05-18:
I'd mentioned this on the forums (but can't find post) - from a user's perspective what I use is "lines" for top/bottom layers, then "top surface skin layers" and "bottom initial pattern" both sent to concentric.

This gives you a nice concentric finish without holes, since the lines one layer beneath doesn't show.

Vandrasc, 2018-06-29:
Fixed in CURA-5276.

closed 2018-06-29 by Vandrasc

midgethoen, 2018-06-29:
Awesome. Thnx