Ultimaker/CuraEngine #744: Gcode corrupted

mogey336, 2018-04-14:
I have a problem with Cura 15.04.6 and Cura 3.2.1
When slicing stl file , the gcode adds delay and the nozzle move outside the model then wait for 1 or 2 seconds then back to the model then continue printing.
The model is correct i tried printing old models i printed before same thing happens!!
I download Cura on my desktop same result also
[image]

diegopradogesto, 2018-04-16:
Can you provide more data? Type of printer, settings, model, the gcode itself, ... If you work with Cura you can also provide the project file.

mogey336, 2018-04-16:
After searching it, i realize i was enabling cool head lift 
That was the proplem

closed 2018-04-16 by mogey336