Ultimaker/CuraEngine #745: [3.3] Support generation issues

jaibooot, 2018-04-21:
**Application Version** : 3.3
**Platform** : windows 64 bit

**Steps to Reproduce**
- import "shallowbottomangle.stl"
- Add interface generation to ultimaker 3 profile
- Support angle set to 45deg
- Observe support generation

**Actual Results**
- when support structures with interface intersect with a base on a shallow angle, the support does not touch the base, and no interface is built 

[image]

- When the model is scaled to 20% of its initial size, support is not generated, but interface is. If supports are not present, interface should also not be generated. 

[image]

**Additional Information**
I have seen this issue in multiple geometries, I just made these samples to better demonstrate what I have been seeing. In addition to these, I have also seen support structures build with no interface on the top or bottom. I have only seen this happen in small overhang areas, typically with a lot of curvature. 

[support_generation_issues.zip](https://github.com/Ultimaker/CuraEngine/files/1935055/support_generation_issues.zip)

jaibooot, 2018-05-02:
The issue still persists with no interface layers. 
[image]

BagelOrb, 2018-05-02:
I was talking about the second case.

I always use the line type coloring in layer view. That really helps to see what lines are support and which ones are model. I got confused for a second.

The support is not resting on the model; it is being pushed away by the model due to the XY distance. If you lower the XY distance the support will rest on the model and the interface will be generated.

I still consider it a bug that it is not automatically generated, though.

BagelOrb, 2018-05-02:
This issue is about the support interface not being generated. If you indeed have multiple issues then please make separate github issues for them.

Ghostkeeper, 2018-05-28:
It's a very difficult problem indeed because what the issue says is that in this case you would want to have some support interface below your support:

[image]

But in this case you don't want interface below your support:

[image]

Locally they are exactly the same, but somewhere next to the support is now a pillar that the support can rest on so then it's fine.

BagelOrb, 2018-05-28:
:+1: good explanation!

jaibooot, 2018-05-29:
I see your point. Interface below the support structure in your second case would be technically unnecessary, but would not have a negative effect. In your first case, the support will not build without the interface layer being printed. Therefore, there is a strong case for building support interface in both cases, even if unnecessary. 

Perhaps, something like conical supports with a - deg value could be of use here? It is optimized for positive values, so it may need some work from the test cases I have used.

Ghostkeeper, 2018-07-10:
Actually I've used conical support more often with negative values than with positive values. It's really useful to have support be wider at the bottom.

ianpaschal, 2018-08-15:
As @BagelOrb said, this is a very difficult issue to solve. Since it can be fixed/alleviated by tweaking some settings, we decided as a team that we can't invest time in trying to make Cura Engine solve it for you. It also makes the engine code unnecessarily complex.

Sorry about that.

closed 2018-08-15 by ianpaschal