Ultimaker/CuraEngine #746: [3.3] Make "regular/maximum" fan speed threshold act on feature size, not layer size

jaibooot, 2018-04-21:
**Application Version** - 3.3

**Actual Results**
- Cura currently applies a higher fan speed when a layer is below a certain time threshold. This works well for something like a cone, where a smaller layer means a smaller feature. Often, there are parts that have layers with large sections of printing and a couple small features. 

example part: 
[image]

Ideally, the fan should be off for the square section, and on for the small pillars. Instead of a minimum layer time, the fan speed threshold would act on a mesh area. One could set the mesh area in mm^2, where the fan should bump up to "maximum speed". The fan would change speed before the travel move to the small section of printing. 

I am not sure how easy this would be to implement, but it would be a big help for a lot of part geometries.

smartavionics, 2018-04-22:
Hi @jaibooot , as I have recently added the ability for the fan to be tweaked while printing bridges and also, most recently, when printing skin areas that are over support, it would be fairly straightforward now to implement what you are requesting in terms of the fan speed control. However, at the moment, having commanded a fan speed change it doesn't wait for the fan to speed up or slow down before the next print operation so if the area to be printed is really small then it would have printed most of it before the fan could speed up. Some thought required there to get that right.

BagelOrb, 2018-05-02:
Why do you need a higher fan speed for small parts?

jaibooot, 2018-05-02:
higher fan speed is helpful for small features. The higher cooling rate keeps the previous layers from softening and deforming,  enduring good feature definition. For larger features (the square in the above example), fan speed is less important and can be decreased.

BagelOrb, 2018-05-02:
The reasoning for the way the cooling works at the moment is to do with the material cooling on itself. If a layer is too small then it doesn't have enough time to cool by itself. We never considered reheating because of the layer we put on top.

I think the functionality we have in place should not be changed.

However, you could request to add the functionality as you want it on top of the functionality as it is in place. It makes things complicated, though; having two such mechanisms interwoven with each other.

jaibooot, 2018-05-07:
would it be possible to add a setting "enforce per feature"? The trigger time and behavior would remain the same, but the increased cooling / head lift would trigger on small features as well as small layers.

jaibooot, 2018-05-29:
Could fan speed be controlled the same way the nozzle heat from standby to print is controlled (inserted preemptively in the gcode)? This would likely also make sense for the fan speed above bridging sections.

BagelOrb, 2018-05-30:
Bridging makes a good use-case for implementing such similar functionality for fan speeds.

I don't really know how bridging is implemented, though.

rickyzhang82, 2018-05-30:
Assuming your fan has PWM pin connected to control board and also your mother board support this, yes you can control fan speed by M code.

IMHO, controlling fan speed per part might be an early optimization. What could go wrong if fan speed is **always** reach high? It make your cool down faster thus it won't deform. I don't know if filament type requiring different cool down speed.

In any case, it is physical problem which requires tons of different variables. It is not that easy to get a formula (algorithm) to automate the tweak in fan speed. I don't have a good idea how to do it in a systematic way so that we can add the logic in a user-friendly way.

Ghostkeeper, 2018-06-07:
> IMHO, controlling fan speed per part might be an early optimization. What could go wrong if fan speed is always reach high?

For PLA you should (almost) always have your fan at 100%, yes. But for ABS or other more warp-y materials, keeping the fan speed high will cause your print to warp if you're printing large objects. You need the material to stay warm for a while longer so that it can anneal, dissipate the tension.

I think increasing the fan speed will have little effect on a small part though, because it usually takes 1 or 2 seconds for the fan to spin up. The part should be almost done by that time. And then the fans don't have a lot of time to cool down the material either.

poon-holder, 2022-09-15:
Has this been implemented elsewhere now?