Ultimaker/CuraEngine #756: Please can we have new Releases for Gentoo packaging

mjeveritt, 2018-05-02:
**Application Version**

**Platform**
Gentoo Linux

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

I have requested via IRC that we'd like full releases rather than simply tags for packaging Cura and its dependencies properly in Gentoo Linux. As they are quite inter-dependent, it would be good to make sure these tie up correctly to enable our users to make best use of the latest editions of the package. Thanks!

ChrisTerBeke, 2018-05-27:
Less than 0.5% of Cura users are on Linux, and only a very small subset of that is on Gentoo I imagine. I'm pretty use that Ultimaker will not work on this, maybe someone in the community will pick it up.

closed 2018-05-27 by ChrisTerBeke

rickyzhang82, 2018-05-27:
@mjeveritt 
It is built in AppImage. Why do you need a specific Linux distribution?

mjeveritt, 2018-06-02:
I'll fork it then, automate pulls & releases based on tags. Tedious, but meh.