Ultimaker/CuraEngine #761: Firmware Retract + Relative Extrusion + Coasting leads to enormous primes

wilriker, 2018-05-07:
**Application Version**
`Cura-3.3.1.AppEngine`

**Platform**
Arch Linux

**Display Driver**
Nouveau

**Steps to Reproduce**
1. Enable Firmware Retract
2. Enable Coasting
3. Enable Relative Extrusion
4. Slice a STL

**Actual Results**
GCode will contain something like this:
```
G11
G1 F1800 E113.40642
```
which means do firmware unretract and (remember relative extrusion) extrude another 113.40642mm of filament right at this spot.

**Expected results**
Amount of priming should be relative.

**Additional information**
The issue will not occur if Coasting is disabled (tested this) and probably not when absolute extrusion is used (I only assume that).

smartavionics, 2018-05-08:
Next major release should see this fixed. Thanks for the nice bug report.

wilriker, 2018-05-08:
@smartavionics Thanks for reacting so quick on this. :-)

smartavionics, 2018-05-08:
You're welcome. I feel honour bound to come up with a fix as I failed to make that change when the relative extrusion feature was added. So it's my bug and I'm going to fix it!

wilriker, 2018-05-08:
Being a software developer myself I can exactly feel what you mean here! :-)

Next major release would be 4.0.0? Or 3.4.0? And is there a timeline?

smartavionics, 2018-05-08:
I should imagine it will be 3.4 but I don't know when - they tend to release rather too often so it probably won't be more than a month or two max.

wilriker, 2018-05-08:
OK, thanks! I will try to get your fix incorporated into my git-master-based installation of CuraEngine meanwhile. :-)

wilriker, 2018-05-08:
Update: I patched my local version of CuraEngine and I can now confirm that your fix kills the bug. Thanks again!

smartavionics, 2018-05-08:
Good, happy printing!

Ghostkeeper, 2018-06-07:
The fix was merged, but not in time for 3.4. Sorry!

closed 2018-06-07 by Ghostkeeper

wilriker, 2018-06-07:
@Ghostkeeper Thanks! And no worries, I am running a patched version of 3.3, so I will also run a patched version of 3.4 - and probably I am one of the few users that will encounter this problem in the first place due to enabling FW Retract not being too easy. ;-)

PR #762 (2018-06-07): Prime amount was incorrect when relative extrusion and firmware retract both enabled.
Fixes #761.