Ultimaker/CuraEngine #777: Use Variable Infill To Prime Nozzle OR Print largest island first?

AbeFM, 2018-05-18:
**Application Version**
3.3.1

**Platform**
N/A

**Qt**
Don't know what QT is.

**Steps to Reproduce**
Print with Dual Extrusion. To exaggerate effect, print one small diameter pole next to a thick one, vary which prints first.

**Actual Results**
I find that I am getting extrusion failures when I'm printing very small thin islands as my first lines after a nozzle change. The larger the infill area for the first island, the better the outer wall looks.

**Expected results**
That the quality of outer wall extrusion is less dependent on the smallest area of a given layer.

**Additional Information**
I would suggest choosing the largest island to start printing. That could be done behind the scenes and would improve prints on ANY machine with non-ideal change-overs (including UM I imagine).
Secondly, one could print a solid infill layer (or as much as needed) in order to prime the nozzle. This will alleviate the need for a priming tower while still providing the benefit.

Probably I would set the density of each layer's infill to use a user-settable "minimum extrusion before outer wall", calculated to use that much filament before any outer wall is printed.
In advanced cases, ALL infill on a given layer could be printed before the first outer wall though I suspect this is a bigger programming job.

----------------------------
I'm printing a series of poles, with variable infill, to try to show this. I'll update as I get results.

Ghostkeeper, 2018-06-29:
Typically we're using the prime tower to fix this. It's not as time-efficient but much simpler to use and doesn't interfere with the desired Layer Start X/Y position.

AbeFM, 2018-06-29:
Several topics regarding prime tower reliability have been.... Also not a
priority.

Could we make some efforts there (brim, solid outer shell of one specific
extruder, etc.)?

When there's trouble with prime tower, don't use it. When you need to do it
somewhere else because tower fails often, it's too hard, just use a tower.

Can we pick one method and look at it seriously? I would like to get multi
material prints looking better.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

AbeFM, 2018-06-29:
It would work to have independent prime towers per nozzle as an option.

In my particular case, I just need a PLA tower, in fact, if I could just
set an object to always print first, I could make my own single material
tower pretty easily.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

alekseisasin, 2018-07-02:
@AbeFM , one of the solutions you can create an empty model like tower and move to the bottom left corner. In the settings change the Layer Start X, Layer Start Y to 0. Now your tower will be printed first

AbeFM, 2018-07-02:
Thanks! I've been doing that.

I don't post on here because MY prints don't work. I post on here in an attempt to improve the product by sharing my insights as a user.

I feel that feature would benefit the community, hence the suggestion. Thanks for yours.