Ultimaker/CuraEngine #784: [Discussion] branching management

rickyzhang82, 2018-05-25:
I'm a conservative on programming. To better manage new features in Cura engine, I make the following suggestion:

1. Split the master branch into stable and experimental. 
1. Every new feature can merge into experimental ASAP.
1. For those features that doesn't impact everyone like adding a new hardware support, you may merge it into stable branch in a timely manner. 
1. But **ONLY and ONLY** cherry pick the well tested feature that impact everyone like tool path generation improvement features from experimental branch into stable branch after a extended long period of test. That test includes user feed back with their physical prints.

In such way, we can keep Cura support more hardware. But we still feel safe to use the latest release from stable branch.

Ghostkeeper, 2018-05-25:
You are suggesting to use the "git-flow" branching model suggested by the well known ["A successful git branching model"](https://nvie.com/posts/a-successful-git-branching-model/) blog post. Instead, we have deliberately chosen the competing ["cactus" branching model](https://barro.github.io/2016/02/a-succesful-git-branching-model-considered-harmful/), which is basically the opposite. We're not about to change that. It was chosen for greater speed of development because it requires fewer merging of large branches. And you're also never merging a large branch into a stable branch, which then causes additional bugs to arise.
Both sides have advantages and disadvantages, of course.

Before a feature gets published in a stable branch though, it gets tested through 3 layers of QA within Ultimaker and then in a public beta as well. I hope that you still feel safe to use the latest release from a stable branch (which is in our case called "3.3" instead of "master").

ianpaschal, 2018-05-25:
Hi @rickyzhang82, @Ghostkeeper pointed me to the topic you started on the Ultimaker forum. I had said make a discussion here because GitHub lacks a good discussion platform when not related to a specific PR or issue. But since we actually _DO_ have a discussion going on there I'm going to close this "issue." I only want to use "discussion issues" on GitHub when there's no other way to have an open ended discussion since there's no clear point at which they're "solved."

In this case, I think @Ghostkeeper's answer is enough of an explanation but we'll keep reading and participating on your forum thread.

closed 2018-05-25 by ianpaschal

Ghostkeeper, 2018-05-26:
All right so let's continue the discussion here then: https://community.ultimaker.com/topic/23059-does-cura-engine-development-go-in-a-wrong-direction/

rickyzhang82, 2018-05-26:
Well, I thought you are a benevolent dictator.  The thread there is way too long and start to off-topic. If you want, we can find a place to have a clean slate to continue our discussion.

ianpaschal, 2018-05-26:
I think it's fine. Better than an issue tracker.

If you want to refocus it on something in particular, Ricky, why not make a post boiling down what you think is the main issue, to bring it back on track? It's true it went in several directions in terms of architecture, development process, and the mixing of open source and for-profit enterprises. I don't want to speak for you but you could summarize this, state which one is the "big issue" in your eyes, and what you'd do to change that.