Ultimaker/CuraEngine #785: [wiki] The term insect in Cura Engine

rickyzhang82, 2018-05-30:
I just deleted the template due to the fact that none of them apply my issue here.

@Ghostkeeper Insect is a term I got troubled to understand when I read 15.04 Cura engine. It took me a while to figure it out what it means. Why do you remove it from the glossary? 

This is nitty gritty. But it is important to save us time for discussion in future.

Ghostkeeper, 2018-05-30:
Insect is not a term in CuraEngine. It's a type of creature: https://en.wikipedia.org/wiki/Insect

Later I was thinking. Did you mean "inset"? Because that is sort of a term, but it's not equivalent to "wall".

rickyzhang82, 2018-05-30:
My bad. But your wiki link just cracked me up....

What is difference between wall and inset? I might misunderstand the term `inset` when reading the source. I do want to know what I missed. TIA.

ChrisTerBeke, 2018-05-30:
wall is the same as inset. inset is just the technical term in the toolpath generation scope.

closed 2018-05-30 by ChrisTerBeke

rickyzhang82, 2018-05-30:
OK. @Ghostkeeper If there is no objection, I will just add 
 - Inset. a.k.a wall.

ianpaschal, 2018-05-30:
I would avoid adding these sorts of "aka" or "see" definitions as they're not really definitions. If we're not sure why the term inset is used, don't fill it in. I'm sure there is a reason the term "inset" was chosen and that's what should be listed there.

rickyzhang82, 2018-05-30:
We do know how inset is used. But @Ghostkeeper doesn't agree with my original proposal that inset = wall. I need his input.

aka and see also is common thing in dictionary. I don't know what other alternative to group all of them together. It is like some say to-mar-to but some say to-make-to. I'm open to your suggestion. But we should not exclude them.

ChrisTerBeke, 2018-05-31:
To be more precise, we use insets to generate wall lines. It's a shrunken version of the outer perimeter of an island. Source: http://alienryderflex.com/polygon_inset/.

rickyzhang82, 2018-06-01:
If I may, I'd define the following way for inset:

Inset is a set of closed polygons generated by slicing engine for the outer perimeter and the inner perimeter (if any) of the island.

ianpaschal, 2018-06-01:
Yes definitely better. As I said, I would avoid "see" or "aka" definitions as it replaces one question "What is an inset?" with a new question "Why not just use the other term then?" Net effect is still having questions.

Pretty much any definition explaining it in more detail, and ideally also how it's related to but different from other, similar terms, is fine by me.

Go for it.

DDDirk, 2018-07-13:
Are there two types of inset then? 
The first is for the outer wall, which is inset half a road width from the outside of the model. This is what I thought inset meant.
The next walls in are inset a full roadwith (if there is no overlap). This is what I understand that you mean.

Ghostkeeper, 2018-08-27:
No. Insets are used for many things. Outer walls and inner walls are just two of them. In our current code base there are 165 calls to the function that performs polygon insets! An inset is **not** a wall.