Ultimaker/CuraEngine #787: After I run make in CuraEngine's build directory, this error comes up.

PrasoonMaheshwari, 2018-06-04:
[  6%] Building CXX object CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o
In file included from /home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/Comb.h:11:0,
                 from /home/ubuntu/workspace/CuraEngine-2.7.0/src/LayerPlan.h:8,
                 from /home/ubuntu/workspace/CuraEngine-2.7.0/src/FffGcodeWriter.h:15,
                 from /home/ubuntu/workspace/CuraEngine-2.7.0/src/FffProcessor.h:5,
                 from /home/ubuntu/workspace/CuraEngine-2.7.0/src/commandSocket.cpp:7:
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(Args ...)’:
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:34:18: error: expected ‘,’ before ‘...’ token
             [args...]()
                  ^
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:34:18: error: expected identifier before ‘...’ token
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:34:21: error: parameter packs not expanded with ‘...’:
             [args...]()
                     ^
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:34:21: note:         ‘args’
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:36:34: error: expansion pattern ‘args’ contains no argument packs
                 return new T(args...);
                                  ^
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(const std::function<_Res(_ArgTypes ...)>&, Args ...)’:
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:52:21: error: expected ‘,’ before ‘...’ token
             [f, args...]()
                     ^
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:52:21: error: expected identifier before ‘...’ token
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:52:24: error: parameter packs not expanded with ‘...’:
             [f, args...]()
                        ^
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:52:24: note:         ‘args’
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:54:36: error: expansion pattern ‘args’ contains no argument packs
                 return new T(f(args...));
                                    ^
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(const std::function<T*(Args ...)>&, Args ...)’:
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:68:21: error: expected ‘,’ before ‘...’ token
             [f, args...]()
                     ^
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:68:21: error: expected identifier before ‘...’ token
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:68:24: error: parameter packs not expanded with ‘...’:
             [f, args...]()
                        ^
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:68:24: note:         ‘args’
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/home/ubuntu/workspace/CuraEngine-2.7.0/src/pathPlanning/../utils/LazyInitialization.h:70:30: error: expansion pattern ‘args’ contains no argument packs
                 return f(args...);
                              ^
make[2]: *** [CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o] Error 1
make[1]: *** [CMakeFiles/_CuraEngine.dir/all] Error 2
make: *** [all] Error 2

[image]

smartavionics, 2018-06-05:
I think that means your gcc is too old. I am current using gcc (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609.

PrasoonMaheshwari, 2018-06-06:
thanks @smartavionics ..it works fine 💯

closed 2018-06-06 by PrasoonMaheshwari