Ultimaker/CuraEngine #788: Trying to retrieve unregistered setting with no value given: 'line_width' kossel plus 230

firegore2, 2018-06-04:
**Application Version**
CuraEngine 3 (lastest in 04/06/2018)

**Platform**
Debian 9
**Steps to reproduce**
```shell

root@firegore:/var/www/caddy/firegore.es/CuraEngine# ./build/CuraEngine slice -v -j printer/kossel_plus_230.def.json -o output/completo.gcode -e0 -l MatM_Iron_Man_Helmet.stl

Cura_SteamEngine version master
Copyright (C) 2018 Ultimaker

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
Loading printer/kossel_plus_230.def.json...
Loading printer/fdmprinter.def.json...
Loading printer/fdmextruder.def.json...
Loading MatM_Iron_Man_Helmet.stl from disk...
[WARNING] Mesh has disconnected faces!
loading 'MatM_Iron_Man_Helmet.stl' took 0.443 seconds
Loaded from disk in 0.470s
[WARNING]  -s center_object="false" -s adaptive_layer_height_threshold="200" -s adaptive_layer_height_variation="0.1" -s wireframe_nozzle_clearance="1" -s wireframe_roof_drag_along="0.8" -s wireframe_roof_fall_down="2" -s wireframe_straight_before_down="20" -s wireframe_strategy="compensate" -s wireframe_top_jump="0.6" -s wireframe_flat_delay="0.1" -s wireframe_flow_flat="100" -s wireframe_printspeed_flat="5" -s wireframe_printspeed_down="5" -s wireframe_printspeed_up="5" -s wireframe_printspeed_bottom="5" -s wireframe_height="3" -s mesh_position_z="0" -s flow_rate_max_extrusion_offset="0" -s magic_fuzzy_skin_thickness="0.3" -s support_conical_angle="30" -s support_conical_enabled="false" -s spaghetti_flow="20" -s spaghetti_max_height="2" -s spaghetti_max_infill_angle="10" -s spaghetti_infill_enabled="false" -s coasting_min_volume="0.8" -s coasting_volume="0.064" -s coasting_enable="false" -s draft_shield_height_limitation="full" -s draft_shield_enabled="false" -s material_flow_dependent_temperature="false" -s roofing_angles="[ ]" -s roofing_pattern="lines" -s roofing_line_width="0.4" -s support_tree_wall_count="1" -s support_tree_collision_resolution="0.4" -s support_tree_branch_distance="4" -s support_tree_angle="40" -s relative_extrusion="false" -s magic_mesh_surface_mode="normal" -s mold_roof_height="0.5" -s mold_enabled="false" -s meshfix_union_all_remove_holes="false" -s meshfix_union_all="true" -s support_tree_enable="false" -s raft_base_fan_speed="0" -s adaptive_layer_height_variation_step="0.01" -s raft_surface_fan_speed="0" -s raft_base_jerk="20" -s raft_interface_acceleration="3000" -s anti_overhang_mesh="false" -s raft_surface_acceleration="3000" -s raft_base_speed="15" -s raft_interface_speed="15" -s raft_surface_speed="20" -s raft_base_line_spacing="1.6" -s raft_base_thickness="0.3" -s raft_interface_line_width="0.7" -s raft_interface_thickness="0.15" -s raft_surface_thickness="0.1" -s support_mesh="false" -s raft_airgap="0.3" -s prime_tower_position_x="200" -s raft_margin="15" -s skirt_gap="3" -s adhesion_type="brim" -s extruder_prime_pos_y="0" -s prime_blob_enable="true" -s support_mesh_drop_down="true" -s infill_mesh_order="0" -s support_minimal_diameter="3" -s remove_empty_first_layers="true" -s support_tower_diameter="3" -s prime_tower_wipe_enabled="true" -s support_use_towers="true" -s support_roof_pattern="concentric" -s support_bottom_line_distance="0.4" -s support_tree_branch_diameter="2" -s support_roof_line_distance="0.4" -s support_interface_skip_height="0.3" -s support_roof_height="1" -s support_bottom_enable="false" -s support_roof_enable="false" -s gradual_support_infill_steps="0" -s mold_width="5" -s support_infill_sparse_thickness="0.1" -s support_join_distance="2" -s support_bottom_stair_step_width="5" -s support_bottom_stair_step_height="0.3" -s support_xy_overrides_z="z_overrides_xy" -s support_xy_distance="0.7" -s wireframe_roof_inset="3" -s dual_pre_wipe="true" -s support_line_distance="2.66" -s support_pattern="zigzag" -s support_angle="50" -s support_type="everywhere" -s support_bottom_extruder_nr="0" -s support_roof_extruder_nr="0" -s infill_hollow="false" -s raft_surface_line_spacing="0.4" -s skirt_line_count="1" -s support_extruder_nr_layer_0="0" -s support_tree_branch_diameter_angle="5" -s support_infill_extruder_nr="0" -s support_skip_some_zags="false" -s raft_interface_fan_speed="0" -s support_wall_count="1" -s support_enable="false" -s cool_lift_head="false" -s cool_min_layer_time="5" -s cool_fan_speed_0="0" -s cool_min_layer_time_fan_speed_max="10" -s cool_fan_speed_max="100" -s wireframe_enabled="false" -s cool_fan_speed_min="100" -s cool_fan_enabled="true" -s wireframe_fall_down="0.5" -s retraction_hop_only_when_collides="false" -s retraction_hop_enabled="false" -s conical_overhang_enabled="false" -s draft_shield_dist="10" -s ironing_pattern="zigzag" -s speed_ironing="20" -s retraction_min_travel="1.5" -s ironing_enabled="false" -s retraction_extrusion_window="4.5" -s ironing_only_highest_layer="false" -s skin_outline_count="1" -s infill_mesh="false" -s print_sequence="all_at_once" -s default_material_print_temperature="210" -s z_seam_y="100" -s xy_offset_layer_0="0" -s material_flow="100" -s skirt_brim_speed="30" -s carve_multiple_volumes="true" -s xy_offset="0" -s jerk_support_bottom="20" -s filter_out_tiny_gaps="true" -s spaghetti_infill_extra_volume="0" -s machine_min_cool_heat_time_window="50" -s support_top_distance="0.1" -s fill_perimeter_gaps="everywhere" -s raft_base_line_width="0.8" -s z_seam_x="100" -s travel_compensate_overlapping_walls_x_enabled="true" -s retraction_hop="1" -s machine_use_extruder_offset_to_offset_coords="true" -s z_seam_corner="z_seam_corner_inner" -s raft_interface_jerk="20" -s skin_line_width="0.4" -s smooth_spiralized_contours="true" -s wall_0_inset="0" -s support_line_width="0.4" -s top_bottom_pattern="lines" -s wall_0_wipe_dist="0.2" -s raft_surface_line_width="0.4" -s wall_line_width_x="0.4" -s wall_line_count="2" -s material_flow_temp_graph="[[3.5,200],[7.0,240]]" -s retraction_hop_after_extruder_switch="true" -s wall_0_extruder_nr="-1" -s machine_gcode_flavor="RepRap (Marlin/Sprinter)" -s prime_tower_line_width="0.4" -s skin_alternate_rotation="false" -s acceleration_enabled="false" -s wall_line_width_0="0.4" -s top_bottom_pattern_0="lines" -s machine_center_is_zero="true" -s retraction_combing="all" -s bottom_layers="6" -s magic_fuzzy_skin_point_dist="0.8" -s retraction_count_max="90" -s cool_fan_full_layer="2" -s speed_travel="120" -s machine_shape="rectangular" -s z_seam_relative="false" -s jerk_skirt_brim="20" -s raft_base_acceleration="3000" -s switch_extruder_prime_speed="20" -s acceleration_infill="3000" -s jerk_infill="20" -s skin_angles="[ ]" -s infill_extruder_nr="-1" -s material_standby_temperature="150" -s acceleration_print_layer_0="3000" -s fill_outline_gaps="false" -s ironing_flow="10" -s machine_nozzle_head_distance="3" -s slicing_tolerance="middle" -s machine_acceleration="4000" -s brim_line_count="20" -s skirt_brim_line_width="0.4" -s infill_line_distance="2" -s jerk_support_infill="20" -s brim_outside_only="true" -s wall_x_extruder_nr="-1" -s spaghetti_inset="0.2" -s support_connect_zigzags="true" -s machine_height="300" -s machine_nozzle_tip_outer_diameter="1" -s magic_spiralize="false" -s ironing_line_spacing="0.1" -s infill_wipe_dist="0.04" -s machine_nozzle_expansion_angle="45" -s machine_max_acceleration_z="100" -s top_bottom_extruder_nr="-1" -s material_diameter="1.75" -s machine_filament_park_distance="16" -s speed_equalize_flow_max="150" -s material_print_temp_wait="true" -s support_conical_min_width="5" -s layer_height_0="0.3" -s machine_nozzle_id="unknown" -s z_seam_type="sharpest_corner" -s machine_name="Unknown" -s support_bottom_height="1" -s machine_max_acceleration_e="10000" -s meshfix_extensive_stitching="false" -s infill_line_width="0.4" -s multiple_mesh_overlap="0.15" -s raft_surface_jerk="20" -s acceleration_travel="5000" -s machine_nozzle_cool_down_speed="2" -s extruder_prime_pos_z="0" -s acceleration_ironing="3000" -s top_layers="8" -s machine_width="250" -s initial_layer_line_width_factor="100" -s jerk_travel_layer_0="20" -s mesh_position_x="0" -s meshfix_keep_o [... cut ...]
Starting export...
Multithreading GcodeLayerThreader with 2 threads.
[ERROR] Trying to retrieve unregistered setting with no value given: 'line_width'
[ERROR] Trying to retrieve unregistered setting with no value given: 'line_width'

```
**Actual Results**
[ERROR] Trying to retrieve unregistered setting with no value given: 'line_width'

**Expected results**
The stl in gcode version in path output/completo.gcode

**Additional Information**

[Stl file and .def.json files](https://firegore.es/files.rar)

smartavionics, 2018-06-05:
This error normally happens when the engine has been rebuilt but the front end has not been updated.

firegore2, 2018-06-05:
What is the front-end? I only use CuraEngine (as it says in the documents)

smartavionics, 2018-06-05:
The front end is the GUI, all that python stuff...

firegore2, 2018-06-05:
hahaha, but in Debian they don't have GUI (at least pre-installed). I want to use CureEngine to create the gcode from a stl, Can it do this?

BagelOrb, 2018-06-06:
This is a bug.

`line_width` was meant to be a pure parent setting. Parent settings should never be retrieved by the engine, because their parent child relationship is interpreted such that if all children are set by the user then the parent becomes unsettable. The parent setting has no meaning other than to control the children. Instead the engine should retrieve a specific line width, e.g. wall_line_width.

smartavionics, 2018-06-06:
It was b85292d348 that added the "line_width" setting usage. Calling @jackha...

commit 559b5897 (2018-06-07): Use more specific line width settings for move inside distance

The width of the line that it needs to cross.

Fixes #788.
closed 2018-06-07 by Ghostkeeper