Ultimaker/CuraEngine #790: [Discussion] How to debug Cura Engine in debugger

rickyzhang82, 2018-06-05:
Pardon me my skipping the template. Because the topic below doesn't relate to bug report nor feature request.

I'm in the middle of adding a feature in Cura Engine in 15.04 branch. I looked into a way to debug it with gdb. 

The input STL file seems to stream from Python UI client through socket with rest of settings passed in from command line. 

I tried to add `gdbserver localhost:5000` before invoking cura engine in Python UI command list. But it failed to launch gdb server. I can't figure it out why Python `subprocess.Popen` behave like this.

Since there is no developer guide in wiki, I'd like to know how you guys as developer debug Cura Engine in Mac or Linux. 

Thanks in advance.

fieldOfView, 2018-06-06:
I can't help you with Cura 15.04.x (because I don't have any experience developing for it, and technically it is not supported here), but I recently found out how to launch CuraEngine in gdb in connection with a running instance of Cura.

* Start Cura with the command-line option --external-backend
```
python3 cura_app.py --external-backend
```
* In the output, look for a line like this:
```
[MainThread] UM.Backend.Backend._createSocket [242]: Listening for backend connections on 49674
```
* Launch CuraEngine manually in gdb, connecting on that port:
```
gdb --args CuraEngine connect 127.0.0.1:49674
```

ChrisTerBeke, 2018-06-06:
That's the development setup I use as well, but then using CLion's debugger and runner.

closed 2018-06-06 by ChrisTerBeke