Ultimaker/CuraEngine #793: Discussion: how can I access JIRA system?

rickyzhang82, 2018-06-11:
I saw Cura system is referenced in PR. I wonder how can I access JIRA as an outsider.

ChrisTerBeke, 2018-06-11:
You can't, there's a bunch of private stuff in there.

closed 2018-06-11 by ChrisTerBeke

rickyzhang82, 2018-06-11:
OK. Is that the way we do open source?

ChrisTerBeke, 2018-06-11:
Yup. Take it or leave it.

ChrisTerBeke, 2018-06-11:
Ultimaker pretty much funds this entire project, so having an internal ticketing system where we define priorities and link to other (closed) software within the organization is perfectly logical. As if Google's open source projects have all their project boards open to show what they are working on.

nallath, 2018-06-11:
Our Jira board also links to hardware / material implementations that are not out yet (and thus sensitive). We've looked into checking how to open this up, but we were unable to do so. Opening it up would also force ultimaker to pay a whole lot more for Jira, as you pay per account.

rickyzhang82, 2018-06-12:
I'm not interested in your proprietary stuffs like firmware, print test on material and etc.

I'm not against your using JIRA in your corporate environment. This is absolute your private business. I respect that.

As an outsider, I will feel that you guys are treating this open source software project like an exclusive fan club. If you really want any outsiders to poke around or help review your PR, we are better off to make the information accessible as much as possible. My suggestion is that you can always refer Github issue or commit in your JIRA but never expose it in Github.

"Take it or leave it" attitude is not quite right. At least, in all open source projects I participated I have never heard such absurd things. Perhaps, it is better off make this closed source if you really want to 'take it or leave it'

ChrisTerBeke, 2018-06-12:
We cannot make it closes source because of the existing license. We also don't intend to as we value all the community contributions, especially for other printers than Ultimaker's. Most information is publicly on GitHub here. The Jira tickets we link to often only contain back-links to GitHub or links to other Jira tickets with private information. All development is done in the open, all branches are open. You can see what we're releasing way before we actually release it. The reason I reacted like that is because we put in an effort and money everyday to maintain this as an open source project, and sometimes it just feels like people are trying their hardest to find something to complain about.

rickyzhang82, 2018-06-12:
As a software developer for a closed source company, I always love to hear customer's complaint. In fact, I valued their opinion. Because that means they do use our software and give us feedback that we as developer overlooked. All developers have such shortcoming that we feel very sensitive when someone criticize our work. It is normal. Because I feel the same way. But once your head cool down, you will realize even the most destructive negative comment has its own constructive side.

We are off topic now. 

My comment on the PR (https://github.com/Ultimaker/CuraEngine/pull/767) have been removed. All I want to do is to poke around this PR under the hood. But you guys quote a JIRA ticket which is not open to any outsider.

If possible, please stop quoting JIRA ticket in Github. Do whatever you want in JIRA by quoting Github issue and ticket.

For QA, I don't think any sane humane can help you. I don't think the QS inside your company to review a PR with tons of commit and no decent documentation in commit message.

Two suggestion I'd propose:
1. When send PR, look into rebase to squash local commit that nobody cares. Minimize the number of commit as many as possible.
1. Write a decent commit message with one line summary and multiple line of details. Your commit message just make me feel you are working like a fresh grad cow boy coder. No offense. But it is extremely difficult for any one want to poke around your PR. Ask your QA. They know this better.

ChrisTerBeke, 2018-06-12:
I don't mind complains/feedback about the actual product, it's just annoying that threads about the process or something else that doesn't matter keep popping up. As for your suggestions, they're something I personally like as well, but no one in the company is doing this so it probably won't be adopted.

rickyzhang82, 2018-06-12:
Well, you still believe this doesn't matter.  What you guys do basically turned away any outsiders. 

This violates the original purpose that makes this project open source in the very beginning. That's why I suggest to keep it closed if you don't need any outsiders.

Anyway, do whatever you want. It is none of my business. Thanks!

ChrisTerBeke, 2018-06-12:
Like I said earlier, these Jira are references are literally just a reminder for us where things are internally. They often only link back to the original issue on GitHub. So believe whatever you want, but Ultimaker is very much committed to keeping this open source, and you can see we adopt many contributions from the community. That we don't adopt your suggestions doesn't say anything about our general ideas of open source contributions, nor does it say anything about your or me personally.

I'm now gonna unsubscribe from this thread cause I'm done with it...

ianpaschal, 2018-06-12:
Key word here is open-**source**. The software and its source code is open. Ultimaker, however, is a for profit company. Facebook has open source projects, but they do not open their entire company infrastructure to the public. Same with Google. Same with literally any company.

It's a non issue and we won't be changing this behavior as it's handy for us when browsing GitHub to know if an internal ticket exists.

Ghostkeeper, 2018-07-23:
I do think that we should not refer to information in Jira from Github. We may link stuff together (e.g. "this pull request is tracked by CURA-1234") but we may not refer to a ticket number without it being clear what that ticket is (e.g. "this will be solved as a side-effect when we implement CURA-4321").

Sophist-UK, 2020-08-31:
I fully understand the issue of Jira having commercially sensitive stuff in it - but if you are going to provide a JIRA ticket code in a Github PR, would it be possible to copy the non-commercially sensitive stiff from JIRA into the Github issue/PR so that contributors can at least see the context (even if it is a one-time snap-shot of what is being discussed internally).

Ghostkeeper, 2020-09-02:
I think it's honestly less useful than you would think :P There's stuffy information like story point estimates, who the stakeholders are or who to keep in the loop, and categorisation tags. There are also requirements in it, which are somewhat useful. Bug reports include reproduction steps, expected and actual behaviour, and some hints for testing/reviewing. I'd say that about 80% of the bugs we're fixing also have a link to where the issue was reported on Github.

We (or at least I) do provide a description of the work that's implemented in the PR, normally, including those requirements and Github links. Just a Jira ticket number is a bit opaque.

Commercially sensitive stuff won't end up in a PR on Github anyway until it's safe to publish (or maybe a few hours ahead). We consider a pull request to be public. For that purpose we have a repository called `cura-private-data` and an alternative track in our build system that pastes the cura-private-repository on top of the source code of Cura. This way we can store profiles and such for developing products in cura-private-data, and have a build that we can distribute within Ultimaker so that the mechanical/process engineers in those teams can work with their developing products.

If we're developing something that we know a certain external developer will be impacted by, we should warn them about it. For instance a while back we were making changes to the profile drop-downs and the settings list, which could impact the Sidebar plug-in, so we warned its author.