Ultimaker/CuraEngine #796: Per-filament first-layer hotend/print temperature

VanessaE, 2018-06-17:
Different filaments don't just need different print temperatures, they need different first-layer temperatures as well.  Just setting a global first-layer temperature via the chosen quality profile is not good enough, as it would require a quality profile for each filament (and perhaps several per, if one has multiple general quality settings they might otherwise wish to use).

One needs to be able to fine-tune that first layer temperature properly per-filament if the bed has something like PEI or Printbite on it -- what works for one filament may be too much adhesion for another, or it might not stick at all.

Compare with Slic3r (it also allows a first-layer bed temperature; I have no use for that, but I know others do).

(this issue opened at @BagelOrb's suggestion :smiley: )

ChrisTerBeke, 2018-06-19:
Although it's probably an engine issue, it's better to post this in the Cura repository as there is a lot more traffic there (also from the core devs).

VanessaE, 2018-06-19:
OOps!  I actually meant to do just that.  

https://github.com/Ultimaker/Cura/issues/3957

closed 2018-06-19 by VanessaE